Xilinx/Vitis_Libraries #2: Building the cornertracker example on xilinx_u200_xdma_201830_2 fails

ThomasXilinx, 2019-10-07:
I tried building the cornertracker example on xilinx_u200_xdma_201830_2 and got a link error:

INFO: [VPL 60-1032] Extracting hardware platform to /home/tbollaer/Work/Labs/Vitis_Libraries/vision/L3/examples/cornertracker/build/_x_xilinx_u200_xdma_201830_2_hw/link/vivado/vpl/.local/hw_platform
[20:30:11] Run vpl: Step create_project: Started
Creating Vivado project.
[20:30:16] Run vpl: Step create_project: Completed
[20:30:16] Run vpl: Step create_bd: Started
[20:31:47] Run vpl: Step create_bd: RUNNING...
[20:32:21] Run vpl: Step create_bd: Failed
[20:32:26] Run vpl: FINISHED. Run Status: create_bd ERROR

===>The following messages were generated while  creating FPGA bitstream. Log file: /home/tbollaer/Work/Labs/Vitis_Libraries/vision/L3/examples/cornertracker/build/_x_xilinx_u200_xdma_201830_2_hw/link/vivado/vpl/vpl.vdi :
ERROR: [VPL-1] /cornerTracker_1/m_axi_gmem1 is associated to /cornerTracker_1/ap_clk on /cornerTracker_1 but this clock pin is not connected to a clock source...
ERROR: [VPL 60-773] In '/home/tbollaer/Work/Labs/Vitis_Libraries/vision/L3/examples/cornertracker/build/_x_xilinx_u200_xdma_201830_2_hw/link/vivado/vpl/runme.log', caught Tcl error:  ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.
ERROR: [VPL 60-773] In '/home/tbollaer/Work/Labs/Vitis_Libraries/vision/L3/examples/cornertracker/build/_x_xilinx_u200_xdma_201830_2_hw/link/vivado/vpl/vivado.log', caught Tcl error:  ERROR: [Common 17-39] 'send_msg_id' failed due to earlier errors.
ERROR: [VPL 60-704] Integration error, Failed to update block diagram in project required for hardware synthesis.The project is 'prj'. The block diagram update script is '.local/dr.bd.tcl'. The block diagram update script was generated by system linker. An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [20:32:30] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:18 ; elapsed = 00:03:23 . Memory (MB): peak = 686.980 ; gain = 0.000 ; free physical = 2333 ; free virtual = 93413
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking

I believe this is coming from the --clock.defaultFreqHz ${VIVADO_FREQUENCY} constraint on line 215 of the Makefile:
https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L3/examples/cornertracker/build/Makefile#L215

I took this out and the run proceeded past the previous point of failure (but isn't finished yet).

xilinx-yang, 2019-10-24:
Hi ThomasXilinx, correct that --clock.defaultFreqHz support from tool was not mature. Additionally there are fixes in our internal code repository already. Once sync internal repository to GitHub, planned this weekend, 10/27/2019 is completed, the build issues should be cleared.

vt-lib-support, 2020-05-22:
issue fixed, close it.

closed 2020-05-26 by vt-lib-support