Xilinx/Vitis_Libraries #3: McEuropeanHestonGreeksEngine data transfer issue

vladimir-eremin, 2019-11-25:
When trying to run an L2 test of McEuropeanHestonGreeksEngine I get zeros for all except the first value:
```
$ make run TARGET=sw_emu DEVICE=u200
export XCL_EMULATION_MODE=sw_emu; \
/home/common/ven_veremin/SDx/Vitis_Libraries-master/quantitative_finance/L2/tests/MCEuropeanHestonGreeksEngine/bin_xilinx_u200_xdma_201830_2/host.exe -xclbin /home/common/ven_veremin/SDx/Vitis_Libraries-master/quantitative_finance/L2/tests/MCEuropeanHestonGreeksEngine/xclbin_xilinx_u200_xdma_201830_2_sw_emu/MCEHGEngine_k0.xclbin
----------------------McEuropeanHestonGreeksEngine-----------------
Found Platform
Platform Name: Xilinx
Found Device=xilinx_u200_xdma_201830_2
INFO: Importing /home/common/ven_veremin/SDx/Vitis_Libraries-master/quantitative_finance/L2/tests/MCEuropeanHestonGreeksEngine/xclbin_xilinx_u200_xdma_201830_2_sw_emu/MCEHGEngine_k0.xclbin
Loading: '/home/common/ven_veremin/SDx/Vitis_Libraries-master/quantitative_finance/L2/tests/MCEuropeanHestonGreeksEngine/xclbin_xilinx_u200_xdma_201830_2_sw_emu/MCEHGEngine_k0.xclbin'
kernel has been created
kernel start------
kernel end------
Execution time 122036782us
difference :
    0: 0.000757404
    1: 0.10813
    2: 0.506724
    3: 1.0021
    4: 4.34831e-06
    5: 0.406697
    6: 3.47461e-06
    7: 0.00260451
theta: -0.0483559, rho: 0, delta: 0, gamma: 0, MV_kappa: 0, MV_theta: 0, MV_KHI: 0, MV_VO: 0
```
If values are printed directly from the kernel in sw_emu they are correct:
```
Kernel results:
-0.048356
-0.108764
0.508515
1.005469
-0.000003
0.407388
0.000424
```
So the issue is obviously with data transfer from kernel to host.

paulc-xlnx, 2019-12-02:
This issue has been identified and resolved.

Fix needs propagated to the branch.

vt-lib-support, 2020-06-17:
issue fixed

closed 2020-06-17 by vt-lib-support