Xilinx/Vitis_Libraries #4: Alveo U200 & U250 - Data center FPGA board platforms

mahinlma, 2020-01-02:
Hi,

For Xilinx Edge device, board platform source given here - https://github.com/Xilinx/Vitis_Embedded_Platform_Source

From where I can get board platforms for Data center cards - Alveo U200 & U250?

mahinlma, 2020-01-03:
@luciferlee , Please provide clarification on this.

songc-xil, 2020-01-06:
Platforms for Alveo U200 and U250 can be found here:
https://www.xilinx.com/products/boards-and-kits/alveo/u200.html#gettingStarted
https://www.xilinx.com/products/boards-and-kits/alveo/u250.html#gettingStarted

mahinlma, 2020-01-07:
@songc-xil , Thank you.
I required "xilinx_u200_xdma_201830_1.xpfm", Please let me know, how to get this.
There is no directory named "Platform" under /opi/xilinx/

mahinlma, 2020-01-07:
@songc-xil 
From where I can get Hardware specification (XSA) for Alveo 200 & 250

songc-xil, 2020-01-07:
Download the "Development Target Platform" from the links I posted, and install them in your system, you should be able to see the XSA for Alveo U200/U250 under /opt/xilinx/..

mahinlma, 2020-01-08:
@songc-xil , Thank you

closed 2020-01-08 by mahinlma

chenruiqiu, 2022-05-21:
hello,i am test the Opencv resize,but i can not find the xpfm of u200，while i have finish these steps in the picture
[image]