Xilinx/Vitis_Libraries #5: SSR FFT example 1dfix_impulse will not synthesize

baileyji, 2020-01-07:
Vitis_Libraries/dsp/L1/examples/1Dfix_impluse requires alteration of hls_ssr_fft_parallel_fft_kernel.hpp before it will synthesize. See https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/Synthesis-hangs-in-HLS-2019-2-1-on-SSR-FFT-example/m-p/1060615 for details.

dylanbespalko, 2020-04-14:
@baileyji,

I am also having this issue and your changes make sense. I still cannot get this example to run-in the hw_emu mode.

How is it possible that the Vitis examples can run without any OpenCL commands communicating between the host and the kernel? Is this entire example supposed to run with the kernel PL fabric?

baileyji, 2020-04-14:
I didn't try using anything with OpenCL. Through some debugging with folks at Xilinx we eventually discovered that their example works without modification when built on linux but seems to just not work on Windows. Period. My guess is that there are compiler issues that I don't have time to track down. In the end we found that the SSR FFT that is in the Matlab System Generator block-set is far more performant, making timing for 16x4096 @ 512MHZ on an Ultrascale+ (ZCU111 eval board) and went that way. 

My hope is that in another month or so we'll see significant improvement with the SSR FFTs as HLx 2020.1 is released.

dylanbespalko, 2020-04-14:
Wow, thanks for the information. I'll look into the Matlab block.

dylanbespalko, 2020-04-16:
Btw, I am running on linux, however I am getting windows related warnings about exceeding the maximum file path length. This could be specific to my installation.

```bash
WARNING: [IP_Flow 19-4832] The IP name 'fft_top_ap_fadd_3_full_dsp_32' you have specified is long. The Windows operating system has path length limitations. It is recommended you use shorter names to reduce the likelihood of issues.
```

gareth-xlnx, 2020-04-16:
> 
> Btw, I am running on linux, however I am getting windows related warnings about exceeding the maximum file path length. This could be specific to my installation.

This is a general caution about using long names rather than any specific limit you are hitting with this instance. It emits these in case you transport the project into a Windows environment at some point in the future - as a Linux user you are safe to ignore.

vt-lib-support, 2020-06-17:
Synthesisability issue has been fixed. working on other functional issues for this design.

vt-lib-support, 2020-09-09:
Issue fixed in latest code.

closed 2020-09-09 by vt-lib-support