Xilinx/Vitis_Libraries #6: Generate "compression_only" and "decompression_only" - xclbin for Lz4

mahinlma, 2020-01-08:
`  make all TARGET=hw DEVICE=xilinx_u200_xdma_201830_2
`
after this hardware compile execution, under ./build/xclbin*/  , found only compress_decompress.xclbin

How to generate bitstream for compress_only & decompress_only, What are the arguments should be enabled to get this? or what modification I should make? Please let me know.

kaliudayxilinx, 2020-01-09:
@mahinlma :  I assume you tried to build at L2 location.

Please build the designs at following location to get compress only and decompress only XCLBINs.

https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L2/tests/lz4_compress
https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L2/tests/lz4_decompress

Also under L3/demos/lz4_app/
By default this app generates two xclbins, compress.xclbin and decompress.xclbin under ./build/xclbin*/  location.

mahinlma, 2020-01-10:
@kaliudayxilinx , Thank you. I will check the mentioned directories.

closed 2020-01-10 by mahinlma