Xilinx/Vitis_Libraries #11: xilLz4CompressStream outputs no more than 4.1KBytes

luojw-dwr, 2020-02-20:
I'm trying to utilize xilLz4CompressStream as part of my kernel. When the input size is large to some extent (I have not  found the certain threshold), the output size will be 4.1KBytes. This behaviour is conformed by putting both a 98KBytes text and its first 49KBytes half but achieving the same two 4.1KBytes outputs (the sameness is conformed with diff).

p.s. To let the core outputs only after receiving some input, I removed the HLS RESOURCE pragmas.

vt-lib-support, 2020-03-16:
@luojw-dwr we have updated the source code recently can you please try and let us know if you still face the same issue.

vt-lib-support, 2020-03-31:
Closing this issue as there is no further update on this. Reopen if this issue still persists.

closed 2020-03-31 by vt-lib-support