Xilinx/Vitis_Libraries #19: CMake set(CMAKE_CXX_COMPILER /path_to/v++) compiler support

dylanbespalko, 2020-05-06:
Hello,

I have added a bunch of the PyTorch kernels [here](https://gitlab.com/pytorch-complex/vitis_kernels) for Vitis.  I was able to get things working using system calls from CMake, but I would appreciate if Vitis would support:
 1. The ability to specify v++ as a c++ compiler using  `set(CMAKE_CXX_COMPILER /path_to/v++)`.
2. The ability to `find_package(Vitis)` within CMake instead of running .sh commands to configure the environment.

**This implicitly needs to be located inside the Xilinx Vitis installation and not Vitis Libraries or Vitis AI**
Related: 

[Vitis AI cmake folder](https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Library/cmake)

[CMake toolchain docs](https://cmake.org/cmake/help/v3.9/manual/cmake-toolchains.7.html?highlight=cmake_c_compiler)

[CMake find_package docs.](https://cmake.org/cmake/help/v3.0/command/find_package.html)

dylanbespalko, 2020-05-06:
cc. @anjn

vt-lib-support, 2020-05-22:
Hi dylanbespalko,

Since this is a Vitis tool related request not for the libraries, could you please post it in Xilinx forum? 
       https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v 

Thanks!

closed 2020-05-22 by vt-lib-support