Xilinx/Vitis_Libraries #22: Relation of "ZIP BOMB" information and maximum compression ratio as fpga::gzip decompress

f1593570, 2020-06-18:
I use std::gzip to compress data and then use fpga::gzip to decompress, but it throw out this information. It shows below : 
```shell
root@Chromium sw_gzip $ xil_gzip -sx /opt/xgzip/compress_decompress.xclbin -d spark3_new.dump.gz
Found Platform
Platform Name: Xilinx
Using Device: xilinx_u200_xdma_201830_2
INFO: Reading /opt/xgzip/compress_decompress.xclbin
Loading: '/opt/xgzip/compress_decompress.xclbin'
KT(MBps)                :

ZIP BOMB: Exceeded output buffer size during decompression

Use -mcr option to increase the maximum compression ratio (Default: 10)

Aborting ....

Segmentation fault (core dumped)
```
And then I check default value of `-mcr` , the default value is 20.
```shell
root@Chromium sw_gzip $ xil_gzip --help

===========================================================
Usage: application.exe, -[-h-c-d-sx-v-dev-l-k-mcr]

          --help,                    -h       Print Help Options         Default: [false]
          --compress,                -c       Compress
          --decompress,              -d       DeCompress
          --single_xclbin,           -sx      Single XCLBIN
          --compress_decompress,     -v       Compress Decompress
          --device,                  -dev     FPGA Card # to be used
          --file_list,               -l       List of Input Files
          --cu,                      -k       CU                         Default: [0]
          --max_cr,                  -mcr     Maximum CR                 Default: [20]
```
But when I set `-mcr 10`, fpga::gzip can decompress successfully.
Is this situation reasonable ?
And how can I use `-mcr` ?

vt-lib-support, 2020-06-19:
@f1593570  what is the compression ratio of the input file when it was compressed? 
**mcr**: means maximum compression ratio, this option is used to allocate larger output buffer (for example -mcr 10 which means we allocated 10 times of input file size (compressed file). If your compression ratio is much higher then you will face this error and -mcr option can be used to create larger output buffer. _In your report it looks like quite opposite happened.  Can you reset the card and try ?

f1593570, 2020-06-24:
@vt-lib-support after I reset FPGA, this situation did not happen again. Thanks for your help.

vt-lib-support, 2020-06-24:
@f1593570 thanks for the update. As Issue is resolved closing it.

closed 2020-06-24 by vt-lib-support

keryell, 2020-06-24:
But it looks like there might be a discrepancy between the default `mcr` value displayed in the help message and the one displayed in the error message.