Xilinx/Vitis_Libraries #23: SSR FFT creates ap_memory instead of ap_fifo

aroshanineshat, 2020-07-05:
Hi, I synthesized 1D SSR FFT and according to the documentation I should get ap_fifo interface but I'm getting ap_memory. 
FFT length is 1024 and SSR factor is 8. 
[Here is the link to Xilinx forum for more detailed explanation.](https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/Vitis-2020-1-1D-SSR-FFT-is-producing-ap-memory-interface-instead/m-p/1124753)

I'm using Vitis 2020.1. HLS  2019.1 produced correct ap_fifo interface.

aroshanineshat, 2020-07-05:
I want to confirm that using the exact same code in **vivado_hls** and activating c++11 flag for gcc, produces the correct ap_fifo interfaces with correct bit count.

vt-lib-support, 2020-07-13:
Hi Arash,

There are some changes on the pragma in Vitis HLS, and it should be the cause of your issue.

In FFT source code, we can find some of the FIFOs are described like:
```
T_scaledFFTOutputType fftOutData_local2[t_R][t_L / t_R];
#pragma HLS data_pack variable = fftOutData_local2
#pragma HLS RESOURCE variable = fftOutData_local2 core = FIFO_LUTRAM
#pragma HLS STREAM variable = fftOutData_local2 depth = 8 dim = 2 
```
In the code above, the third pragma is no longer supported in `vitis_hls`, so it is ignored and consequently leads to an `ap_memory` interface because `vitis_hls`  now sees `fftOutData_local2` as a buffer instead of a FIFO. 

It is no longer the preferred way to use pragma to specify an array as FIFO, and the code will be refactored to use `hls::stream` when a FIFO is expected.  _We will fix Vitis HLS compatibility issues in DSP sub library in 2020.2 release._ 

The current release should work with **`vivado_hls`**, although the tool has entered end of its life cycle.

closed 2020-08-24 by vt-lib-support