Xilinx/Vitis_Libraries #25: vitis vision opencv version?

DavideConficconi, 2020-07-31:
I am having issues in compiling the vision examples on a CentOS 7-based machine with Vitis 2020.1 by using the Makefile provided.

I do get as error what follows
```
xf_warp_transform_tb.cpp:104: undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
xf_warp_transform_tb.cpp:121: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
xf_warp_transform_tb.cpp:121: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
```

I am trying with OpenCV4.4 module.
Is this an issue related to my OpenCV version, or is it something else?

Thank you in advance

vt-lib-support, 2020-09-21:
Hello @DavideConficconi,

Vitis Vision functions have been tested against OpenCV 3.4.2 version.

Please make sure to correctly set the OPENCV_INCLUDE and OPENCV_LIB environment variables, before executing the Makefile.

vt-lib-support, 2020-09-21:
Hello @DavideConficconi please check with the suggested env settings.

closed 2020-09-21 by vt-lib-support

3togo, 2020-10-21:
@DavideConficconi , have you fixed the compiling issue?
I got very similar problem even after I have exported the  OPENCV_INCLUDE and OPENCV_LIB to opencv3 path

/home/eli/git/Vitis_Libraries/vision/L1/tests/sgbm/v64_16_300/sgbm.prj/sol1/csim/build/../../../../../../../examples/sgbm/xf_sgbm_tb.cpp:384: undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
/home/eli/git/Vitis_Libraries/vision/L1/tests/sgbm/v64_16_300/sgbm.prj/sol1/csim/build/../../../../../../../examples/sgbm/xf_sgbm_tb.cpp:385: undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
/home/eli/git/Vitis_Libraries/vision/L1/tests/sgbm/v64_16_300/sgbm.prj/sol1/csim/build/../../../../../../../examples/sgbm/xf_sgbm_tb.cpp:438: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::

DavideConficconi, 2020-10-22:
@3togo honestly I did not test back yet. Which opencv version are you using?

DavideConficconi, 2020-10-23:
@3togo  I am currently testing the lib, the warpaffine example hardware builds for the Vitis 2019.2 flow, and I also tested as a standalone in Vivado HLS (working on host test)

Maybe have a look at the README for the 2020 version (i.e., master branch currently) [here](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L2/README.md).
Did you setup also LD_LIBRARY_PATH?