Xilinx/Vitis_Libraries #26: bug in xf::cv::AXIvideo2xfMat

Ali-Flt, 2020-08-24:
Hi, 
I'm trying to use xf::cv::AXIvideo2xfMat and there seem to be a bug inside the code causing the synthesizer to error.
the error is as followed:
```
ERROR: [HLS 200-977] Argument 'stream_in_1_V' failed dataflow checking: it can only be used in one process.
WARNING: [HLS 200-992] Argument 'stream_in_1_V' has read operations in process function 'Loop_loop_wait_for_start_proc26'.
Resolution: For help on HLS 200-992 see www.xilinx.com/html_docs/xilinx2020_1/hls-guidance/200-992.html
WARNING: [HLS 200-992] Argument 'stream_in_1_V' has read operations in process function 'Loop_loop_height_proc'.
Resolution: For help on HLS 200-992 see www.xilinx.com/html_docs/xilinx2020_1/hls-guidance/200-992.html
```
I think it is because the stream input is being read in two different loops in the function code in xf_infra.hpp file. (loop_wait_for_start and loop_height)

I also believe that the 
can you please fix this issue cause I really need this function for my work.

thanks
Ali

canonant, 2020-08-29:
I meet the same problem, could this bug be fixed as soon as possible?

Ali-Flt, 2020-08-31:
After waiting for the Xilinx Employees for a fix and not receiving even a comment, I've tried to fix the function myself. I tested it in Csim and in hardware implementation and it seems to be working. What I basically did was merging the loop_wait_for_start and loop_height  in a single loop. I've attached the modified "xf_infra.hpp" file. 

[xf_infra.zip](https://github.com/Xilinx/Vitis_Libraries/files/5151865/xf_infra.zip)

canonant, 2020-09-01:
@Ali-Flt thanks for you answer, after test you code, I can build this function with "dataflow". but I met some question after load as hardware,[https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/K2K-AXIvideo2xfMat-not-worked/td-p/1145308](url), did you met this problem.

Ali-Flt, 2020-09-01:
Hi @canonant ,
Unfortunately I have never worked with SDAccel and have no Idea what your problem might be, But I will share the code of my Core and C Test bench, in which I used this function, and maybe it will give you an idea about what the issue might be.
I doubt that your problem is from the AXIvideo2xfMat because I tested my modified function in design and in worked properly in Linux Userspace.
[simple_erosion.zip](https://github.com/Xilinx/Vitis_Libraries/files/5156824/simple_erosion.zip)

vt-lib-support, 2020-09-03:
Hi @Ali-Flt / @canonant 
We have reproduced the issue and we are validating the fix we

vt-lib-support, 2020-09-16:
Issue fixed.

closed 2020-09-16 by vt-lib-support