Xilinx/Vitis_Libraries #28: Issue with common/xf_axi_sdata.hpp file

Ali-Flt, 2020-09-17:
Hi, 
I've noticed that when I use the structs defined in the **"xf_axi_sdata.hpp"** file in the Interfaces of my IP core, the HLS synthesis tool doesn't synthesis the interface properly. But when instead I use the same structs from the **"ap_axi_sdata.h"** file it works properly, I cannot see any difference in the codes of these structs but somehow the ones from xf_axi_sdata.hpp don't get synthesized correctly.
Here is an example of a synthesized axis interface using the xf_axi_sdata.hpp file structs:
[image]
Here is the same interface using the ap_axi_sdata.h file structs:
[image]

(the interface must have a 192 bit data width and a 1 bit tuser, tid and tdest data width)

vt-lib-support, 2020-09-21:
Hello @Ali-Flt,

Can you please provide any testcase so that we can further investigate?

Ali-Flt, 2020-09-21:
Hello @vt-lib-support ,
Sure thing, I've attached an example core. The function of the core is to apply erosion on an input stream of data and send it to an output steam. 
As you can see in line 9 of **"wrapper.hpp"** file, I've included the **<ap_axi_sdata.h>** file. Additionally I've commented the **#include "xf_axi_sdata.hpp"** line in the /common/xf_infra.hpp file and added **#include <ap_axi_sdata.h>** instead. So that the structs in this file use the code in the correct file. 
Now if you do the opposite and include **"xf_axi_sdata.hpp"** instead of **<ap_axi_sdata.h>** in both wrapper.hpp and xf_infra.hpp,  you will see that the RTL Ports and number of their Bits in the Interface section of the Synthesis details Report change to false values, just like the values in the image I showed in my previous comment. 
[erosion_example.zip](https://github.com/Xilinx/Vitis_Libraries/files/5253743/erosion_example.zip)

millertheripper, 2020-11-30:
I can confirm the described behaviour with current development master branch. Using <ap_axi_sdatah.h> brings the desired axi stream ports while xf_axi_sdata.hpp does not. 

I can understand why Xilinx currently inverts a lot of resources in the conversion of the HLS libraries to AXI memory mapped accesses, because it is easier and more universal to use with an OS. Nevertheless I think that the AXI streaming interfaces play an important role, because they use exactly the properties of an FPGA (pixel streaming). I really hope that Xilinx will fix this problem soon.

vt-lib-support, 2020-12-23:
@Ali-Flt , @millertheripper 
**xf_axi_sdata.hpp** file has been removed and **ap_axi_sdata.h** is used wherever necessary, in the current released code.
But, co-simulation doesn't pass because of a known Vitis-HLS issue. It is being resolved and a code update will be posted soon. Thanks.

millertheripper, 2021-01-28:
Thank you. I can confirm that the issue with the axi stream parameters is now solved. But in fact the co-simulation fails. Do you know if this is an issue related to the vitis libraries or is it caused by the tool itself (Vivado/Vitis 2020.1)?

Regards

vt-lib-support, 2021-02-18:
Co-sim issue has been fixed with the recent code update.

vt-lib-support, 2021-02-18:
Please reopen if you find any more issues with this function.

closed 2021-02-18 by vt-lib-support