Xilinx/Vitis_Libraries #29: Issue with  structs file

yashwanthresham, 2020-09-17:
Hello! I stuck with this error

../home/l/vitis2020.1/Vitis_Libraries-master/vision/L1/include/common/xf_structs.hpp:778:44: error: 'xf::cv::Mat<3, 1080, 1920, 1>::DATATYPE' {aka 'struct ap_uint<32>'} has no member named 'chnl'
778 | data[r * packcols + c].chnl[p][ch] = in_val;

Is this an issue related to inclusion of headers ? or something else?

Thanks in advance!

vt-lib-support, 2020-09-21:
Hello @yashwanthresham,

Can you try by adding the compile time flag "-D__SDSVHLS__"?

yashwanthresham, 2020-09-23:
yeah if i  done the same i am getting the error as shown, 

\include\CL/opencl.hpp:800:15: error: 'cl_int' does not name a type
 static inline cl_int errHandler (cl_int err, const char * errStr = NULL)
               ^~~~~~

Do we require to install opencl to run this examples? im new to this vitis hls .

vt-lib-support, 2020-10-01:
Yes, OpenCL is required to run L2/L3 examples.
Please go through the [Vitis Application Acceleration Development Flow Documentation](https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/kme1569523964461.html) for all the prerequisites and general understanding about Vitis.

For Vitis Vision specific prerequisites and details, this is the documentation:
https://xilinx.github.io/Vitis_Libraries/vision/2020.1/index.html

vt-lib-support, 2020-12-23:
Please post in [Xilinx Forums](https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/bd-p/tools_v) for further help on environment setup and library usage.

closed 2020-12-23 by vt-lib-support