Xilinx/Vitis_Libraries #35: Xilinx Tree Swaption Engine HW Model Benchmark testing

ymoulayr, 2020-11-09:
I tried to test the following benchmark template from the Vitis Quantitative Finance libraries.
Xilinx Tree Swaption Engine HW Model Benchmark (see attached screenshot) 
[image]

When running the SW build I have the following errors (screenshot attached)
[image]

Could you please advise how the project should be created/configured to manage the dependencies?

ymoulayr, 2020-11-09:
Attached the zip archives of the test_lib and test_lib_system project folders and Emulation-SW logs
[test_lib_Emulation-SW.build.ui.log](https://github.com/Xilinx/Vitis_Libraries/files/5512803/test_lib_Emulation-SW.build.ui.log)
[xcd.log](https://github.com/Xilinx/Vitis_Libraries/files/5512804/xcd.log)
[test_lib.zip](https://github.com/Xilinx/Vitis_Libraries/files/5512796/test_lib.zip)
[test_lib_system.zip](https://github.com/Xilinx/Vitis_Libraries/files/5512799/test_lib_system.zip)

vt-lib-support, 2020-11-12:
Hello, based on the vitis_hls.log for the kernel, your system may be missing some packages. Can you confirm which OS you are using?
Thanks

ymoulayr, 2020-11-12:
Hello,
Thank you for looking into this issue. I am using Linux Ubuntu 18.04.2 LTS.
Please let me know which packages I should install.
Thanks

ymoulayr, 2020-11-12:
Hello,Thank you for looking into this issue. I am using Linux Ubuntu 18.04.2 LTS. 
Please let me know which packages I should install.Thanks

vt-lib-support, 2020-11-12:
Can you install libc6-dev-amd64 and see if that helps?

ymoulayr, 2020-11-12:
Thanks, after installing libc6-dev-amd64 and libc6-dev-x32, I have now the
following error (see attached log):

INFO: [v++ 60-1306] Additional information associated with this v++ link
can be found at:
Reports:
/home/ymoulayr/workspace/vitis/test_lib/Emulation-SW/scanTreeKernel.build/reports/link
Log files:
/home/ymoulayr/workspace/vitis/test_lib/Emulation-SW/scanTreeKernel.build/logs/link
...
INFO: [v++ 60-629] Linking for software emulation target
INFO: [v++ 60-423]   Target device: xilinx_u50_gen3x16_xdma_201920_3
ERROR: [v++ 60-1139] Failed to parse --slr option: The SLR ID string
specified as SLR2 is not valid. The valid SLR IDs are: SLR0, SLR1
ERROR: [v++ 60-703] Failed to finish linking

Could you please advise how to address it?
Thanks

vt-lib-support, 2020-11-12:
Looks like you are targetting a U50, however the benchmark, as indicated in the Readme.md, was designed to run on the U250. It is unlikely you will be able to use the Makefile and settings 'as-is' on the U50. You will likely need to migrate the project.

vt-lib-support, 2020-12-17:
Please follow [these steps](https://gist.github.com/vt-lib-support/0f5f2bdf00d3e7baa50c83c79845dfbe) to run this benchmark on U50 in GUI.

Essentially, these steps replaces the JSON file which defines GUI project to target U50.

closed 2021-02-25 by vt-lib-support