Xilinx/Vitis_Libraries #36: Gzip decompression error with Alveo U50

dajovu2020, 2020-11-17:
Hi,

We have tried to use Gzip decompression/compression with Alveo U50.
Here are the errors when use the default test and our customized test data.

- Hardware platform: Alveo U50
- OS: Ubuntu 18.04
- XRT: xrt_202010.2.7.766_18.04-amd64-xrt.deb

1) make run TARGET=hw DEVICE=u50
    Error message as below.
[image]

2) We use the test data as the below website.
     http://sun.aei.polsl.pl/~sdeor/index.php?page=silesi
    The compression of all data are good but there are errors when decompression.
    Error message as below.
[image]

vt-lib-support, 2020-11-17:
@dajovu2020  Please try gzip demo located at L3/demos/gzip_hbm this demo supports U50.
Link: https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L3/demos/gzip_hbm

dajovu2020, 2020-11-20:
[image]
Hi, 
We have tried L3 but there are the other errors message as the attachment.
Are there anything that we can to modify?
Thanks.

vt-lib-support, 2020-11-21:
Hi @dajovu2020  Please re-run the build and ensure your machine is dedicated for only this.

vt-lib-support, 2020-12-14:
@dajovu2020  since there is no more update on this item. Closing this issue in case if you face trouble reopen or create a new one with logs.

closed 2020-12-14 by vt-lib-support