Xilinx/Vitis_Libraries #39: error while building the rtm2d example from the Vitis Libraries

PaoloPalazzari, 2020-12-17:
while building the rtm2d example, taken from the Vitis hpc libraries, I got an error.

I use the vitis 2020.2 version and from the VITIS_LIBRARY_FOLDER/hpc/L2/tests/rtm2d/rtm/ folder I gave the following command 

make build TARGET=hw DEVICE=xilinx_u280_xdma_201920_3

after a while, I got the following error:

INFO: [Project 1-461] DRC finished with 1395 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

Time (s): cpu = 00:01:02 ; elapsed = 00:00:40 . Memory (MB): peak = 10738.539 ; gain = 0.000 ; free physical = 23673 ; free virtual = 81946
INFO: [Common 17-83] Releasing license: Implementation
172 Infos, 503 Warnings, 7 Critical Warnings and 101 Errors encountered.
opt_design failed
opt_design: Time (s): cpu = 00:01:03 ; elapsed = 00:00:41 . Memory (MB): peak = 10738.539 ; gain = 0.000 ; free physical = 23673 ; free virtual = 81946
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.

INFO: [Common 17-206] Exiting Vivado at Tue Dec 15 14:43:10 2020...
[Tue Dec 15 14:43:11 2020] impl_1 finished
WARNING: [Vivado 12-8222] Failed run(s) : 'impl_1'
wait_on_run: Time (s): cpu = 00:00:14 ; elapsed = 00:23:16 . Memory (MB): peak = 5724.684 ; gain = 0.000 ; free physical = 23690 ; free virtual = 81967
INFO: [OCL_UTIL] internal step: log_generated_reports for implementation 'output/generated_reports.log'
INFO: [OCL_UTIL] internal step: problem implementing dynamic region, impl_1: opt_design ERROR
INFO: [OCL_UTIL] status: fail (opt_design ERROR)
INFO: [OCL_UTIL] log: /home/palazza/Vitis_Libraries/hpc/L2/tests/rtm2d/rtm/build_dir.hw.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.runs/impl_1/runme.log
ERROR: caught error: problem implementing dynamic region, impl_1: opt_design ERROR, please look at the run log file '/home/palazza/Vitis_Libraries/hpc/L2/tests/rtm2d/rtm/build_dir.hw.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
[14:43:25] Run vpl: Step impl: Failed
INFO: [OCL_UTIL] current step: vpl.impl failed. To rerun the existing project please use --from_step vpl.impl
problem implementing dynamic region, impl_1: opt_design ERROR, please look at the run log file '/home/palazza/Vitis_Libraries/hpc/L2/tests/rtm2d/rtm/build_dir.hw.xilinx_u280_xdma_201920_3/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
INFO: [Common 17-206] Exiting Vivado at Tue Dec 15 14:43:25 2020...

In order to help the debug, I attach some reports taken from the build directory[
[VitisError.zip](https://github.com/Xilinx/Vitis_Libraries/files/5709259/VitisError.zip)
](url)

liang-xilinx, 2020-12-17:
Hi @PaoloPalazzari, would you mind building it again in the directory hpc/L2/tests/rtm2d/rtm/tests/dataset_h128_w128_t10_s1/ to see whether it encounters the same error in hw build?

PaoloPalazzari, 2020-12-17:
Hi Liang
  I ran the build on the project you pointed me to, and all was smooth. I
attach the logs of the build, where you can see that there were no errors.
Thank you for your support

  Regards

  Paolo

Paolo Palazzari, Ph.D.
ENEA C.R. Casaccia
TERIN-ICT-HPC
tel (+39) 06 3048 3167
mobile (+39) 393 90 43 433

liang-xilinx, 2020-12-18:
Hi @PaoloPalazzari ,

It's great to hear you have it built successfully. In fact, the projects under the two directories are almost identical except the NO. processing elements. 
Would you mind building the HW again in the directory hpc/L2/tests/rtm2d/rtm/  within the same shell or with identical shell environment? Sorry for the troubles and making you build it again and again. 
In addition, I build the HW under hpc/L2/tests/rtm2d/rtm/ successfully on my local machine. Of course, I will double check it on different machines or environments.  

P.S. I don't see the report in the attachment.

PaoloPalazzari, 2020-12-18:
Just started the build. I will inform you about its progress.
Regards
Paolo

PS : I attach again the tar of the reports of the previous build, hoping
that you can see them

Paolo Palazzari, Ph.D.
ENEA C.R. Casaccia
TERIN-ICT-HPC
tel (+39) 06 3048 3167
mobile (+39) 393 90 43 433

PaoloPalazzari, 2020-12-18:
Hi Liang,
  the build finished in error.
In the following lines, I show you the folder from where I launched the
build and the build command.

palazza@cresco-xilinx02 rtm>pwd
/home/palazza/Vitis_Libraries/hpc/L2/tests/rtm2d/rtm

palazza@cresco-xilinx02 rtm>make build TARGET=hw
DEVICE=xilinx_u280_xdma_201920_3

The log of the build is in the attachment.
Regards

Paolo

Paolo Palazzari, Ph.D.
ENEA C.R. Casaccia
TERIN-ICT-HPC
tel (+39) 06 3048 3167
mobile (+39) 393 90 43 433

vt-lib-support, 2021-02-25:
Hi @PaoloPalazzari, are you still running into this issue?

PaoloPalazzari, 2021-02-26:
Hi,
  currently, I am not trying these. I am now doing some other basic tests
to characterize the behavior of the board. In the near future, I plan to
run again the rtm and, should still I encounter problems, I'll let you know.
  Best regards
Paolo Palazzari

Paolo Palazzari, Ph.D.
ENEA C.R. Casaccia
TERIN-ICT-HPC
tel (+39) 06 3048 3167
mobile (+39) 393 90 43 433