Xilinx/Vitis_Libraries #42: Vitis_Libraries/vision/L2/ Makefile does unnecessary check for embedded targets

mkycrb, 2021-01-22:
Instructions at the following README specify env-vars for PCIe vs embedded:

- [https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2#commands-to-run](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2#commands-to-run)

But when attempting to make for an embedded target the Makefile fails due to the PCIe env-vars missing.  Tested with https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L2/examples/resize/Makefile but maybe applies to other examples as well.

Workaround was simple:

export OPENCV_INCLUDE=0
export OPENCV_LIB=0

vt-lib-support, 2021-02-18:
Hi @mkycrb ,

We acknowledge this limitation with our current Makefiles. We will get rid of this unnecessary setting for embedded platforms in the next release.

vt-lib-support, 2021-08-17:
Removed the dependence on variables OPENCV_INCLUDE and OPENCV_LIB in the makefiles for embedded platforms

closed 2021-08-17 by vt-lib-support