Xilinx/Vitis_Libraries #43: Makefile 2020.1

aniltirli, 2021-01-23:
If embedded XPLATFORM selection is different than  zcu102 and zcu104, there is no LDFLAGS assigned. Could embedded platforms be more generically specified within makefile? 

Makefile lines 127-148

`ifeq ($(HOST_ARCH), x86)
LDFLAGS += -L$(XILINX_VIVADO)/lnx64/tools/fpo_v7_0 -Wl,--as-needed -lgmp -lmpfr -lIp_floating_point_v7_0_bitacc_cmodel
endif
ifneq (,$(shell echo $(XPLATFORM) | awk '/u200/'))
CXXFLAGS += -I$(OPENCV_INCLUDE)
LDFLAGS += -L$(OPENCV_LIB)
else ifneq (,$(shell echo $(XPLATFORM) | awk '/zcu102/'))
CXXFLAGS += -I$(SYSROOT)/usr/include
CXXFLAGS += --sysroot=$(SYSROOT)
LDFLAGS += -L$(SYSROOT)/usr/lib
LDFLAGS += -L${SYSROOT}/opt/xilinx/xrt/lib
else ifneq (,$(shell echo $(XPLATFORM) | awk '/u50/'))
CXXFLAGS += -I$(OPENCV_INCLUDE)
LDFLAGS += -L$(OPENCV_LIB)
else ifneq (,$(shell echo $(XPLATFORM) | awk '/zcu104/'))
CXXFLAGS += -I$(SYSROOT)/usr/include
CXXFLAGS += --sysroot=$(SYSROOT)
LDFLAGS += -L$(SYSROOT)/usr/lib
LDFLAGS += -L${SYSROOT}/opt/xilinx/xrt/lib
endif
`

vt-lib-support, 2021-02-18:
Hi @aniltirli ,

We will try to have the flag settings generic in next release.

vt-lib-support, 2021-02-25:
@aniltirli 

I am going to close this out for now. If during the next release you are still running into issues with the makefile we can revisit.

Thanks

closed 2021-02-25 by vt-lib-support