Xilinx/Vitis_Libraries #44: How to use vitis libraries on ZCU102

JingchiZhang, 2021-01-26:
I noticed that all vitis libraries can only use on Alveo U200 or U250. How can I use them on ZCU102? Is there any documents I can refer to?

vt-lib-support, 2021-02-18:
Hi @JingchiZhang ,

Vitis Vision functions, part of Vitis Libraries, are well supported for embedded platforms like ZCU102 and ZCU104. You may refer to the README files in [Vision ](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision) L2/L3 folders, that list the required settings and commands to run the example testcases on embedded targets.

vt-lib-support, 2021-02-25:
Hi @JingchiZhang 

Have you been able to utilize the libraries with the ZCU102?

In general, you can adjust the Makefiles to target a ZCU102 based Vitis platform. Prebuilt platforms for this board are available on xilinx.com

I'll go ahead and close this issue out, but please do not hesitate to open another one if you need additional help. The Xilinx forums are also another great resource for support.

closed 2021-02-25 by vt-lib-support