Xilinx/Vitis_Libraries #45: How to use database library

BiscuitsColonel, 2021-01-29:
Hi.
The vitis database library is currently being compiled successfully.
After that, I'm thinking of connecting the vitis database library with PostgreSQL to offload SQL processing to the FPGA.
However, it seems that the mechanism is not provided.
I checked your company's github at this URL and it seems that this is the mechanism that connects postgreSQL and FPGA.
Can this library be used with the vitis database library?

vt-lib-support, 2021-01-29:
Hi, 

Organizing storage and optimizing execution plan are heavily studied topics in database domain, and thus are not where we can contribute effectively. Instead, the Vitis Database Library focuses on accelerating most-time consuming execution step(s) common to all SQL engines. We collaborate with some ISV's SQL engine team as following:

```
                          +
         DB Library       |      SQL Engine
                          |
                          |
                 +---+    |   +-------------------+
+------------+   | G <--------> Columnar storage  |
| GQE kernel <---> Q |    |   +-------------------+
+------------+   | E |    |
                 |   |    |   +-------------------+
+------------+   | C <--------> Execution planner |
| GQE kernel <---> + |    |   +-------------------+
+------------+   | + |    |
                 |   |    |
+------------+   | A |    |
| GQE kernel <---> P |    |
+------------+   | I |    |
                 +---+    |
                          +
```

Vitis Database Library now employs a fundamental assumption that _only useful column should be offloaded to PCIE acceleration card_. Unfortunately, PG's default storage engine is not a columnar, though some open-source projects are exploring this direction.

The XPG project you mentioned is an early attempt in enabling PG acceleration, with limited execution plan support. Vitis Database Library is not designed to work with it, though some L1 module can be re-used to create/update new XPG kernel.

BiscuitsColonel, 2021-01-29:
Thank you for your prompt response.

I will ask you a new question based on your answer.
In the source file of Vitis database library, there is [a source file](https://github.com/Xilinx/Vitis_Libraries/tree/master/database/L3/tests/gqe/join/host) for executing TPC-H query No. 5 on FPGA.
How should this file be handled?
Below is information on the environment I am currently verifying.
```
root@Vitis-Test:~# cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
root@Vitis-Test:~#
root@Vitis-Test:~# echo -e "XILINX_XRT : $XILINX_XRT\nPATH : $PATH\nLD_LIBRARY_PATH : $LD_LIBRARY_PATH\nPYTHONPATH : $XILINX_XRT/python:$PYTHONPATH"
XILINX_XRT : /opt/xilinx/xrt
PATH : /insight/xilinx/Vitis_HLS/2020.2/bin:/insight/xilinx/Model_Composer/2020.2/bin:/insight/xilinx/Vitis/2020.2/bin:/insight/xilinx/Vitis/2020.2/gnu/microblaze/lin/bin:/insight/xilinx/Vitis/2020.2/gnu/arm/lin/bin:/insight/xilinx/Vitis/2020.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/insight/xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/insight/xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/insight/xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin:/insight/xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-none/bin:/insight/xilinx/Vitis/2020.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/insight/xilinx/Vitis/2020.2/tps/lnx64/cmake-3.3.2/bin:/insight/xilinx/Vitis/2020.2/aietools/bin:/insight/xilinx/Vivado/2020.2/bin:/insight/xilinx/DocNav:/opt/xilinx/xrt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/home/user1/bin:/usr/lib/postgresql/9.6/bin
LD_LIBRARY_PATH : /opt/xilinx/xrt/lib:
PYTHONPATH : /opt/xilinx/xrt/python:/opt/xilinx/xrt/python:
root@Vitis-Test:~#
root@Vitis-Test:~# initdb -V
initdb (PostgreSQL) 9.6.20
```

vt-lib-support, 2021-01-29:
Hi,

This is the test case that verifies and demonstrates our _join_ API by performing joining of biggest two tables involved in Q5. The case is self-contained, and does not actually connects to any real SQL engine. 

The case creates datasets by itself, in [the `prepare` call](https://github.com/Xilinx/Vitis_Libraries/blob/master/database/L3/tests/gqe/join/host/test_q5s.cpp#L129)  a list of commands are executed to download tool, compile tool, and generate TPC-H data. This binary then loads data to memory as plain column buffers and invokes the GQE API.

This case runs just like all other Vitis-based projects in the library, through `makefile`. Please refer to [design flow page of HTML doc](https://xilinx.github.io/Vitis_Libraries/database/2020.2/design_flows.html) for details. In short, you will
1. source Vitis toolchain setup script
2. run `make` in the folder.

BiscuitsColonel, 2021-01-29:
Thank you for your reply.

In the answer, I understood that you took the benchmark without using "real SQL engine".
So if I want to use the database library through the "real SQL engine", what steps should I take?
If possible, could you please give me an example of the implementation procedure based on the environmental information I collaborated with?
If you need the necessary information, we will additionally cooperate.

vt-lib-support, 2021-01-29:
Hi,
From a high-level view, the follow items are necessary to use GQE Overlay APIs:

1. Choose a column-oriented engine;
2. Modify the engine's query compiler/optimizer to generate calls to Overlay APIs.
3. Enable the engine to assemble plain integer column buffers to call Overlay APIs.

These steps are tightly coupled with implementation of specific engine, so you may need to look for details in the engine's developer document.

BiscuitsColonel, 2021-01-29:
My question has been resolved.
Thank you very much.

closed 2021-01-29 by BiscuitsColonel