Xilinx/Vitis_Libraries #47: How to use REVERSE FFT?

zy486at189cn, 2021-02-24:
When I use REVERSE FFT on zcu102,System crash. But use FORWARD FFT ,it success , why? This is the code

#include "vt_fft.hpp"
#define FFT_LEN 4096
#define SSR 4
#define IN_WL 18
#define IN_IL 14
#define TW_WL 18
#define TW_IL 14
#define IID 0

using namespace xf::dsp::fft;
typedef std::complex<ap_fixed<IN_WL, IN_IL> > T_in;

// Define parameter structure for FFT
struct fftParams : ssr_fft_default_params {
    static const int N = FFT_LEN;
    static const int R = SSR;
    static const scaling_mode_enum scaling_mode = SSR_FFT_NO_SCALING;
    static const fft_output_order_enum output_data_order = SSR_FFT_NATURAL;
    static const int twiddle_table_word_length = TW_WL;
    static const int twiddle_table_intger_part_length = TW_IL;
    //static const transform_direction_enum transform_direction = FORWARD_TRANSFORM;
    static const transform_direction_enum transform_direction = REVERSE_TRANSFORM;
};
typedef ssr_fft_output_type<fftParams, T_in>::t_ssr_fft_out T_out;

T_in p_inData[SSR][FFT_LEN / SSR];
T_out p_outData[SSR][FFT_LEN / SSR];
    for (int r = 0; r < SSR; ++r)
    {
        for (int t = 0; t < FFT_LEN / SSR; ++t) {
            p_inData[r][t] = std::complex<ap_fixed<IN_WL, IN_IL> >(4,4);
       }
    }

 fft<fftParams>(p_inData,p_outData);

vt-lib-support, 2021-02-25:
Hi, I've reached to the library developer for more information on this, please give us a few days to respond.

vt-lib-support, 2021-03-04:
Hi, we have observed an issue of co-sim stall while running reverse FFT, and we are working on a fix.

zy486at189cn, 2021-04-03:
@vt-lib-support   Have you fixed it?

vt-lib-support, 2021-04-19:
Hi, We have fixed the hang issue, and will push-out the fix soon. Stay tuned.

zy486at189cn, 2021-08-02:
@vt-lib-support When will  you fix it?

halalboro, 2025-01-23:
Hi @vt-lib-support, 

Has this issue with the REVERSE_TRANSFORM been fixed? I am using the 2022.1 branch. 

Thanks.