Xilinx/Vitis_Libraries #48: U250 support query

durgabhavaniv, 2021-03-01:
[](https://github.com/Xilinx/Vitis_Libraries/tree/master/database/L1/tests/hw/bitonic_sort)

Does alveo U250 support HW run for the above example?

vt-lib-support, 2021-03-01:
Hi,

This is an L1 API, meaning that it defines a building block of kernel. So as long as the device has enough resources, the API can be implemented on that card. For this L1 API, we are sure it can be configured to fit in U250.

Meanwhile, if you are looking for a ready-to-use Vitis kernel, please take a look at ["Compound Sort"](https://github.com/Xilinx/Vitis_Libraries/tree/master/database/L1/benchmarks/compound_sort) project, which supports U250.

durgabhavaniv, 2021-03-01:
hi @vt-lib-support 

Thanks for a quick reply.

Under Vitis project not able to view test folder.

[https://github.com/Xilinx/Vitis_Libraries/tree/master/database/L1/tests](url)

vt-lib-support, 2021-03-01:
Hi,

As all L1 APIs are HLS modules and do not have proper interface to be used as Vitis kernels, we implemented their unit tests in HLS. Currently, Vitis and Vitis HLS have independent GUI, so these case are not shown in Vitis list.

You can follow [the design flow](https://xilinx.github.io/Vitis_Libraries/database/2020.2/design_flows.html#hls-cases-command-line-flow) to create HLS projects:
```
cd /L1/tests/anyone
make run DEVICE=/path/to/xpfm CSIM=1 COSIM=1
```

These commands will setup and run an HLS project. If you like to design in GUI, you can fire Vitis HLS, and re-open the project created in command line flow above, to see the source, run simulation, and try changes.

durgabhavaniv, 2021-03-01:
@vt-lib-support 

Thanks for the input.

closed 2021-03-01 by durgabhavaniv