Xilinx/Vitis_Libraries #50: How to change the value of VBNV?

BiscuitsColonel, 2021-03-01:
Hi,

I am testing a database library using two FPGA cards.
Specifically, it is the following test.

```
root@Vitis-Test:/insight/xilinx/Vitis_Libraries/database/L3/tests/gqe/aggregate/build_dir.hw.xilinx_u280_xdma_201920_3# ./test_q1.exe -xclbin /insight/xilinx/Vitis_Libraries/database/L3/build/vt_database_aggr-xilinx_u280_xdma_201920_3.xclbin
--------------- Query 1, aggregation ---------------
dat dir is: db_data//dat1
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
Read Input table form db_data//dat1
Using StrategyManualSet
Select solution:2
Lineitem SF(1)  6001215 rows
Result table    1000 rows
LineItem table has been read from disk
INFO: selected device 0: xilinx_u280-es1_xdma_201910_1
INFO: initilized context.
INFO: initilized command queue.
XRT build version: 2.8.743
Build hash: 77d5484b5c4daa691a7f78235053fb036829b1e9
Build date: 2020-11-16 00:19:11
Git branch: 2020.2
PID: 114286
UID: 0
[Mon Mar  1 13:15:36 2021 GMT]
HOST: Vitis-Test
EXE: /insight/xilinx/Vitis_Libraries/database/L3/tests/gqe/aggregate/build_dir.hw.xilinx_u280_xdma_201920_3/test_q1.exe
[XRT] ERROR: Xclbin does not match shell on card.
[XRT] ERROR: Shell VBNV is 'xilinx_u280-es1_xdma_201910_1'
[XRT] ERROR: Xclbin VBNV is 'xilinx_u280_xdma_201920_3'
[XRT] ERROR: Use 'xbmgmt flash' to update shell.
[XRT] ERROR: See dmesg log for details. err=-95
[XRT] ERROR: Failed to load xclbin.
ERROR: fail to program PL
```

As you can see, the shell being called is different and the test cannot be executed.
Could you please tell me how to change the value of VBNV?

vt-lib-support, 2021-03-02:
Hi,

You are compiling to U280 release shell while trying to run the binary on ES version of U280, that's why the xclbin gets rejected.

You would need _a platform compatible with both U280 ES1 and Vitis 2020.2_ first, and then try compile the xclbin with the right platform.

Best,
Yuanjie

BiscuitsColonel, 2021-03-02:
Ty

closed 2021-03-02 by BiscuitsColonel