Xilinx/Vitis_Libraries #51: L1/examples/SGBM hls转换后资源过大，且和api-reference差距较大问题

fanfanba, 2021-03-01:
使用vitis_hls 2020.2对sgbm进行rtl转换，结果和api-reference.rst中给出的参考资源差别很大。
使用的tcl脚本：
https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L1/examples/sgbm/run_hls.tcl
同时参考api-reference.rst中，将xf_config_params.h修改成如下：
            
            #define HEIGHT 1080//720
            #define WIDTH 1920//1280
            
            /* set penalties for SGM */
            #define SMALL_PENALTY 20
            #define LARGE_PENALTY 40
            
            /* Census transform window size */
            #define WINDOW_SIZE 5
            
            /* NO_OF_DISPARITIES must be greater than '0' and less than the image width */
            #define TOTAL_DISPARITY 64
            
            /* NO_OF_DISPARITIES must not be lesser than PARALLEL_UNITS and NO_OF_DISPARITIES/PARALLEL_UNITS must be a
             * non-fractional number */
            #define PARALLEL_UNITS 32
            
            /* Number of directions */
            #define NUM_DIR 4

hls输出的报告如下：

            + Timing: 
                * Summary: 
                +--------+---------+----------+------------+
                |  Clock |  Target | Estimated| Uncertainty|
                +--------+---------+----------+------------+
                |ap_clk  |  3.30 ns|  8.010 ns|     0.89 ns|
                +--------+---------+----------+------------+
            
            + Latency: 
                * Summary: 
                +---------+---------+-----------+-----------+---------+---------+----------+
                |  Latency (cycles) |   Latency (absolute)  |      Interval     | Pipeline |
                |   min   |   max   |    min    |    max    |   min   |   max   |   Type   |
                +---------+---------+-----------+-----------+---------+---------+----------+
                |  2088739|  8668900|  16.731 ms|  69.438 ms|  2088724|  8668880|  dataflow|
                +---------+---------+-----------+-----------+---------+---------+----------+
            
            ================================================================
            == Utilization Estimates
            ================================================================
            * Summary: 
            +-----------------+---------+------+--------+--------+-----+
            |       Name      | BRAM_18K|  DSP |   FF   |   LUT  | URAM|
            +-----------------+---------+------+--------+--------+-----+
            |DSP              |        -|     -|       -|       -|    -|
            |Expression       |        -|     -|       0|      28|    -|
            |FIFO             |        -|     -|    1782|    1206|    -|
            |Instance         |      211|   134|   31051|   68952|    -|
            |Memory           |        -|     -|       -|       -|    -|
            |Multiplexer      |        -|     -|       -|      45|    -|
            |Register         |        -|     -|       5|       -|    -|
            +-----------------+---------+------+--------+--------+-----+
            |Total            |      211|   134|   32838|   70231|    0|
            +-----------------+---------+------+--------+--------+-----+
            |Available        |     1824|  2520|  548160|  274080|    0|
            +-----------------+---------+------+--------+--------+-----+
            |Utilization (%)  |       11|     5|       5|      25|    0|
            +-----------------+---------+------+--------+--------+-----+

api-reference.rst 中给出的latency和资源报告如下：

+----------------+----------------+---------------------------+------------------+-----------+-------+-------+
| Operating Mode | Filter Size    | Operating Frequency (MHz) |           Utilization Estimate               |
+                +                +                                +------------------+-----------+-------+-------+
|                |                |                                    | BRAM_18K         | DSP_48Es  | FF    | LUT   |
+================+================+===========================+==================+===========+=======+=======+
| 1 Pixel        | 5x5            | 200                       | 205              | 141       | 11856 | 19102 |
+----------------+----------------+---------------------------+------------------+-----------+-------+-------+
.. rubric:: Performance Estimate
The following table summarizes a performance estimate for a 1920x1080
image.
.. table:: Table . SemiGlobalBM Function Performance Estimate Summary
   +-------------+-------------+-------------+-------------+-------------+
   | Operating   | Operating   | Number of   | Parallel    | Latency     |
   | Mode        | Frequency   | Disparities | Units       |             |
   +=============+=============+=============+=============+=============+
   | 1           | 200 MHz     | 64          | 32          | 42 ms       |
   | pixel/clock |             |             |             |             |
   +-------------+-------------+-------------+-------------+-------------+

可以看到，api-reference中给出的LUTs只有19k，而我这里转换出的LUTs有70k请问问题在哪里？如何进行优化，降低资源？
谢谢

vt-lib-support, 2021-03-02:
Hi @fanfanba ,

The resource numbers mentioned in the [documentation](https://xilinx.github.io/Vitis_Libraries/vision/2020.2/api-reference.html#semi-global-method-for-stereo-disparity-estimation) are:

- Not from the synthesis report, but the final numbers after completing the Vivado Place and Route stage.
- Only correspond to the Kernel, excluding the data converter functions i.e., Array2xfMat and xfMat2Array.

Please run the 'Export RTL' for your project, which can be enabled by passing 'VIVADO_IMPL=1' argument for your _make_ command. Once it finishes, open the created _.xpr_ file in Vivado tool, open the implemented design and select _Report Utilization_ . This shows the resource numbers of individual blocks of the design in a hierarchy. From the list, find the resource numbers for the kernel function (in this case SemiGlobalBM).

fanfanba, 2021-03-03:
已复现，谢谢

closed 2021-03-03 by fanfanba