Xilinx/Vitis_Libraries #52: Question about Vitis SPARSE library

pmpakos, 2021-03-01:
Hello. I am studying the Sparse (CSC) implementation of Vitis Accelerated Libraries.

From [here](https://xilinx.github.io/Vitis_Libraries/sparse/2020.2/user_guide/L2_intro.html),  I understand that for Alveo U280, we want the 2 **readWriteHbm** CUs need to be placed in SLR0, so that they have direct connectivity with the available HBM channels.

I have also read, that we want the Compute Units that access HBM channels to reside in the SLR that is directly connected with them, therefore SLR0 in Alveo U280 (and Alveo U50 too).

However, reading the configuration files [here](https://github.com/Xilinx/Vitis_Libraries/blob/master/sparse/L2/tests/cscmv/conn_u280.cfg#L140), I noticed that **readWriteHbm_0** is placed in **SLR0** and **readWriteHbm_1** is placed in **SLR1**. Is there any reason that this is written like this? I understand that you want each **readWriteHbm**  to be "close" to the **cscRow** CUs that are assigned to them, but in this configuration there is an SLR crossing (readWriteHbm_1 -> HBM Channels), that, from what I have read, we want to avoid as much as possible.

What of the 2 is correct? Put both readWriteHbms in SLR0, close to HBM channels, or it doesn't make much of a difference to place them elsewhere ?

Thank you!

lisaliu1, 2021-03-01:
Hi, your understanding is right. The idea is to put readWriteHbm kernel close to the HBM banks, which reside in SLR0. However, there is not enough resource in SLR0 to put readWriteHbm kernel and cscRowKernel. Hence, in the configuration file, the second readWriteHbm kernel is positioned in SLR1. This design just demostrate the flexibility with configuring kernel's location when you use kernel-to-kernel streaming design.

pmpakos, 2021-03-01:
Ok, thank you for the really quick answer! 
I will test this, but just asking, in case you already tested it. Could you place all cscRow CUs in SLR1 and both readWriteHbm CUs in SLR0? Or this would be suboptimal from performance perspective, even if there are enough resources to support such a design?

lisaliu1, 2021-03-01:
I think I tried this. That would be suboptimal in terms of achievable clock frequncy. The reason might be when you place readWriteHBM in SLR1, you can leverage dedicated SLR cross for HBM controllers.

pmpakos, 2021-03-01:
Ok, makes sense. Thank you very much!

closed 2021-03-01 by pmpakos