Xilinx/Vitis_Libraries #53: Cannot build Sparse example (for Alveo U50 - for 16 Compute Units) in Vitis GUI

pmpakos, 2021-03-05:
Hello. I want to ask something about the SPARSE library. You have 2 demos in L2/tests directory, **cscmv** and **cscmvSingleHbm**. The first targets U50 and U280 platforms, whereas the second U280 only.
The second one can be used as a template for testing in Vitis GUI, while the first one cannot.

I managed to build the cscmv example from terminal and run it on an Alveo U50 without a problem. However, I want to do everything through Vitis, so that I can utilize run_summary and other capabilities that Vitis GUI offers.

I made some modifications and created a copy of **cscmvSingleHbm** folder (named **cscmv_test** - you can find it [here](https://drive.google.com/drive/folders/1SdUu5RoNMCTdfxBNyW3R_tUf6fIzyApG?usp=sharing)), and modified description.json so that it will be visible in Vitis, and several other modifications so that it can use 16 Compute Units and be equivalent to the **cscmv** example (I also copied Makefile, utils.mk and {*}.cfg from **cscmv** folder) 

However, when I try to build **sw_emu** version, it fails during linking, with the fail message being 
**_ERROR: [v++ 17-1309] Gcc: /opt/xilinx/Vitis/2020.2/bin/../data/emulation/include/xcl_top_defines.h:9:‘packetType {aka struct ap_uint<128>}’ has no member named ‘data’_**

In addition, after trying to build **hw_emu** version, the fail message now is : 
**_ERROR: [CFGEN 83-2284] No stream resources found that can accomodate compute unit "loadColKernel_0.out0"_**

Sorry if this is not the right place to ask this question, but can you wonder what I am doing wrong ?

lisaliu1, 2021-03-05:
hi, some of the features are not supported yet in GUI. Please use command line.

pmpakos, 2021-03-05:
If it's not too much of a hassle, could you please create a demo that will support Alveo U50 in GUI, when you release an update of Vitis Libraries?  Thank you!

closed 2021-03-24 by pmpakos