Xilinx/Vitis_Libraries #54: Cannot run sw_emu on my computer (sparse cscmv test)

pmpakos, 2021-03-08:
Hello. Usually, I use Vitis to develop a project locally on my computer, where I run sw_emu checks to check if everything works properly, before running hw build, that takes much time, on another computer, that is connected to an Alveo U50 board.
Therefore, I have no problem using Vitis sw_emu, despite not having an Alveo board connected to my local computer.

I am trying to do the same thing for the Sparse->L2->tests->cscmv project, first run sw_emu on my local computer, then build it on remote computer. However, even though build does not fail somewhere, when I try to "make run TARGET=sw_emu", all I get is the error message "**Error calling err = cl::Platform::get(&platforms), error code is: -1001**". What am I doing wrong ?

lisaliu1, 2021-03-08:
hi, Vitis sw_emu doesn't support massive streaming connections as shown in cscmv yet.

pmpakos, 2021-03-08:
Since you are always asking to my questions, do you also have any idea why I can't run sw_emu or hw_emu in a docker container? Not only for this example, but generally for every project.. (I know that docker+vitis is not supported/endorsed by you). Sorry if this is not a relevant question, but this question has not been answered in the Xilinx forum, where I have asked also.

lisaliu1, 2021-03-08:
hmm, that's not really my expertise.

pmpakos, 2021-03-08:
Ok thank you, no problem. Final question/issue! You told me that sw_emu does not support massive streaming, but the cscmv example, when compiled with sw_emu target, produces the cscmv_seq (sequential - only one Compute Unit), so it should work, right ?

lisaliu1, 2021-03-08:
haha, you got it. Yes, that's correct. cscmv_seq, a sequential version, only contains one compute unit, works with sw_emu.

pmpakos, 2021-03-08:
thank you very much! (surely, i will bother you in the future again, bye for now)

closed 2021-03-08 by pmpakos