Xilinx/Vitis_Libraries #55: cscmv test does not run for all matrices

pmpakos, 2021-03-10:
Hello. I am running the test of Sparse Library on an Alveo U50. At first, I ran it with the example matrices that are placed in mtxFiles directory. Later, I wanted to test it on larger matrices too (with nonzeros ranging from 100K - 1M). However, for several matrices, cscmv gets stuck at phase 
```
**Loading: '.../cscmv.xclbin'**
```

lisaliu1, 2021-03-10:
Hi, 
For some matrices, the buffer size in the hardware is not big enough to store the intermediate data. If we allocate bigger buffer on hw, we reduce the achievable fmax. The same thing will happen on Alveo U280. The idea is to identify a set of sparse matrices you want to run on FPGA, and we can configure the best possible buffer size for that.

pmpakos, 2021-03-10:
Is there a guide on how to configure the buffer sizes, i.e. will I be able to find it by searching through the code, or you want to keep this feature for your own development purposes? 
My second question is, do I have to reset the card after every matrix that I test? Because sometimes I notice that if I do not, execution gets stuck for matrices that otherwise run successfully (i.e. for matrices that run if at first i "xbutil reset")
Final question (for now) is regarding the "API_Perf(GFlops/Sec)" metric that is provided. What is the main difference with "KernelPerf(GFlops/Sec)" ? Kernel regards computation time only, while API includes the data transfer (in and out) time also ?

Thank you very much for the support!

lisaliu1, 2021-03-10:
Hi, for the time being we would like to keep the configuration as part of the development process because of the potential routing issue involved. If possible, can you paste here a set of matrices you are interested in and we will try to see if we can create a successful .xclbin for them. 
In terms of resetting the card, it's best to reset it after each run at the time being. This issue normally happens when you had a run that failed before. If your run was successful, you should not need to reset the card. 
The API time includes the kernel triggering time, the KernelPerf is the pure hardware run time.

pmpakos, 2021-03-10:
Nice. Thank you once again for immediate response! I will come back with the set of matrices

pmpakos, 2021-03-22:
Hello once again!

I think that I found what you referred to as "internal buffers" and their limited size. I believe that you meant the 
**```l_rowStore[t_MaxRowBlocks][t_Latency]```** buffer that is created in **_rowMemAcc_** hw function. 
The reason that you created this buffer to accumulate partial results along the row indices. The size of t_Latency is set at 6, so that it complies with the Initation Interval of the computations loop that is minimized at 6 cycles.
Please correct me if I am wrong in anything until now.

For the matrices that could not run on this design, I assume that the reason that for some matrices the design gets stuck is that there are more than 6 intermediate results to accumulate per row block. Therefore, I decided to set **t_Latency** to 9. This design was synthesized with no problem and I ran it on Alveo U50.
For the matrices that I had already ran successfully, I noticed a slight drop in performance (6% on average).
For the matrices that did not run previously, not even one of them ran successfully now. So what is the problem with these matrices?

Attached [here](https://docs.google.com/spreadsheets/d/1g4OKglT6QyifaVZWiy7IGLXfuE8Sg-y34cYvidBTTX0/edit?usp=sharing) you will find info about the matrices that I ran (_MM_good_) and did not run (_MM_bad_). I have uploaded the matrices [here](https://www.dropbox.com/sh/6248b4wzrm6lu6t/AABDpCnGHPHaFVpmZFn0V7EKa?dl=0) so that you access them instantly.

Thank you for your help once again,
Panagiotis

lisaliu1, 2021-03-22:
Hi, thanks a million for looking into this. I think it's more related to the buffer for storing dense vectors. In our coming release, we implemented a better version of this SPMV engine, also supoort float64 data type, and showing you how to use it in a solver. So, I will look at the matrices you gave me. But you should get a better version in next release:).

pmpakos, 2021-03-23:
Sorry for interrupting again. But I have one issue and one question : 
I will start with the issue. I ran the (successful) matrices for 10, 100, 1000, 10000, 100000 iterations and noticed differences in performance. To become more specific, the "evolution" of KernelPerf(GFlops/Sec) was (respective to iterations presented previously) :

11.412509 | 14.222098 | 14.959471 | 4.227205 | 0.291877

While for 100 and 1000 iterations it is expected to improve performance, compared to 10 iterations, when iterations become larger (10000, 100000), why do we see such a great drop in performance? How do you explain it?

Regarding the question, I cannot understand what the differences between **SPARSE_maxColParBlocks** and **SPARSE_maxColMemBlocks** are. I understand that the first one is used to partition the matrix in 2D Blocks of **SPARSE_maxColParBlocks*16** columns (and **SPARSE_maxRowBlocks*4** rows). But what about the second?
Finally, could you also explain two me what the compile-time parameters **SPARSE_maxParamDdrBlocks** and **SPARSE_maxParamHbmBlocks** mean?

Thanks :)

lisaliu1, 2021-03-24:
Hi, it's really great that you seem to want to understand the design not just use it. I'll look into the design in more detail today since it has been a while since I looked at it, and get back to you later.

lisaliu1, 2021-03-24:
Hi, here are the answers to your questions
1. the significant drop of perf you obeserved might be caused by increased read/write access conflict to the same HBM channel. The readWriteHbm kernel accesses the same HBM channel for reading the input NNZ & row indices data, and writing the partial result data. The read and write commands will significantly increase with the increase of iterations, hence more conflicts for HBM access.
2. You are right, SPARSE_maxColParBlocks and SPARSE_maxRowBlocs are used to decide the 2D partition size, which has SPARSE_maxRowBlocks * 4 rows, and SPARSE_maxColParBlocks*4 columns. The SPARSE_maxColMemBlocks decides how many memory words required for storing SPARSE_maxColParBlocks*4 dense input vector entries, it's also used to decide the on-chip buffer size for the dense input vector. So, if you use HBM to store your dense input vector as you did when you built your design on U50, the SPARSE_maxColMemBlocks = SPARSE_maxColParBlocks*4 / (256bits/32bits) = 512*4/8 = 256. So, you should change the SPARSE_maxColMemBlocks = 128 in param_u50.mk to SPARSE_maxColMemBlocks = 256
3. the SPARSE_maxParamDdrBlocks is used to decide maximum memory blocks used to store partition paramemter. The SPARSE_maxParamHbmBlocks is used to decide the maximum memory blocks used to store row blocks parameters. The total number of row blocks = number of rows / (SPARSE_maxRowBlocks * 4), and each row block needs 4 32-bit parameters, or 1 HBM block, SPARSE_maxParamHbmBlocks = maximum row blocks. The number of 2D parititions = rows * cols / partition_size, each 2D partition needs 34 p32-bit parameters, or 5 HBM blocks. So, SPARSE_maxParamDdrBlocks/5 = maximum 2D partitions

lisaliu1, 2021-03-24:
The design was originally targetting U280 and uses DDR to store the input dense vector and 2D partition parameters, that's why it has DDR in the definitions. As you can see, the shortcoming of this design is mixing the storage of parameters and data. It maximize the usage of the storage, but makes it hard to maintain and being used in other design. Our next release will not have this isse.

pmpakos, 2021-04-21:
Hello again! 

I would like to ask you if I should change the configuration parameters in Alveo U280 configuration file.
[params_u280.txt](https://github.com/Xilinx/Vitis_Libraries/files/6348766/params_u280.txt) I attach it here, in case it helps

As far as U50 is concerned, I changed SPARSE_maxColMemBlocks to 256 (from 128) as you suggested. I also changed SPARSE_maxColParBlocks to 1024 (from 512), to keep the ratio between them at 4. Is this wrong?
[params_u50.txt](https://github.com/Xilinx/Vitis_Libraries/files/6348771/params_u50.txt) I attach this here too, in case it helps.

Finally, do you have any news on when the updated sparse library will be pushed to repo? Thank you :)

lisaliu1, 2021-04-21:
hi,
I don't see any change in your params_u280.txt compared to the original one. As for the param_u50.txt, you should follow the equation below:
SPARSE_maxColMemBlocks = SPARSE_maxColParBlocks*4 / (256bits/32bits)

The SpMV here is not supposed to be used from SW as an accelerator alone. It's supposed to be used in a bigger accelerator, e.g. linear solvers with high precision support (float64). We re-designed SpMV to support float64 and address the buffering requirement of bigger matrices as you discovered in the current version. We also show you how to use the SpMV in a bigger accelerator. All these will be available in our next release, which will come soon.

pmpakos, 2021-04-21:
Ok, thank you for the u50 equation explanation.
Should this equation be applied to param_u280 too ? Because in this case I should also change SPARSE_maxColMemBlocks to 256 for the param_u280 ( and keep SPARSE_maxColParBlocks to 512).
If not, what is the differentiating factor?

lisaliu1, 2021-04-21:
no, because on U280, the column vector / input dense vector X is store in DDR, and DDR controller has 512bits. Therefore, the equation for U280 becomes:
SPARSE_maxColMemBlocks = SPARSE_maxColParBlocks*4 / (512bits/32bits)
so, you don't need to change U280 parameter configuration.

pmpakos, 2021-04-21:
Thank you very much! Now (almost) everything makes sense!

pmpakos, 2021-04-22:
Hello again. I compiled the version of U50 with the changes that you proposed ( SPARSE_maxColMemBlocks = 256, SPARSE_maxColParBlocks = 512). However, it gets stuck during the "loading xclbin" phase.

Do you have any idea why it gets stuck? I cannot explain it somehow.

When I tried it with the "wrong" version that I had originally assumed  ( SPARSE_maxColMemBlocks = 256, SPARSE_maxColParBlocks = 1024 ), it ran for all matrices that I used as input (regular diagonal matrices like this one that I created) [sparse_2048_block_2.mtx.txt](https://github.com/Xilinx/Vitis_Libraries/files/6356640/sparse_2048_block_2.mtx.txt)

( I had also mentioned it in the past, that the cscmv_test gets stuck for many matrices. In most cases I just ignore it and try another matrix. However, now that the matrices that were used are "regular", and they had ran successfully in the "wrong" version of U50, maybe something bad happens setting the ColMem and ColPar parameters the way that equation shows.. Keep it in mind for U50 configuration file of the future )

Once again, thank you for your attention

lisaliu1, 2021-04-22:
Let me try to re-produce your issue here and give you an update soon.

lisaliu1, 2021-04-22:
hi, just built the .xclbin with following parameters for u20, and the .xclbin works properly with the example .mtx file. No issue with downloading .xclbin as you seen.
SPARSE_maxParamDdrBlocks=128
SPARSE_maxParamHbmBlocks=64
SPARSE_paramOffset=1024
SPARSE_maxColMemBlocks = 256
SPARSE_maxColParBlocks = 512
SPARSE_maxRowBlocks = 512
SPARSE_dataType = float
SPARSE_indexType = uint32_t
SPARSE_logParEntries = 2
SPARSE_parEntries = 4
SPARSE_logParGroups = 0
SPARSE_parGroups = 1
SPARSE_dataBits = 32
SPARSE_indexBits = 32
SPARSE_hbmMemBits = 256
SPARSE_ddrMemBits = 256
SPARSE_hbmChannels = 16
SPARSE_hbmChannelMegaBytes = 256
SPARSE_printWidth = 6
SPARSE_pageSize = 4096
DEBUG_dumpData=0
SEQ_KERNEL=0

The DEVICE used to build this .xclbin is: xilinx_u50_gen3x16_xdma_201920_3

lisaliu1, 2021-04-22:
hi, tried your sparse_2048_block_2.mtx, it also works fine.

pmpakos, 2021-04-23:
Unfortunately, this version does not run for me, for either of 2 matrices. Do you have any idea why? Should I install everything about the card from scratch? (xrt  + deployment shell + development shell)

lisaliu1, 2021-04-30:
Hi,
Can you provide me a .txt file that includes all URLs of the matrices that failed your testing? Something like a test.txt file, which contains a list of URLs similar to the one below:
https://suitesparse-collection-website.herokuapp.com/MM/Simon/raefsky4.tar.gz

Thanks

pmpakos, 2021-05-04:
Hello Lisa. Sorry, i had been offline for a few days. Here are two lists of matrices that I would like to get results.

The first one is matrices that ran successfully on U280 and U50 (U50 list is a subset of U280, so you can ignore it maybe)
[successful_runs_on_Alveos.txt](https://github.com/Xilinx/Vitis_Libraries/files/6421133/successful_runs_on_Alveos.txt)

The second one is the list of matrices that failed to run on U280 
[unsuccessful_runs_on_Alveos.txt](https://github.com/Xilinx/Vitis_Libraries/files/6421146/unsuccessful_runs_on_Alveos.txt)

These are matrices that contain from 230K to 4M nonzeros (apart from last one that contains 11M)
If you would like any further explanations, ask me without hesitation! Thank you

lisaliu1, 2021-05-06:
Hi Pmpakos,
Thanks for the lists. I am happy to let you know that all matrices on your both lists pass the test for our double precision SpMV support. You will get this implementation in our coming release and be able to get the benchmark results easily.

pmpakos, 2021-05-06:
This is great news! Looking forward to it. Thank you.

pmpakos, 2021-06-04:
Hello!
I would like to ask whether the update of sparse library is coming in the near future (next days or several weeks)!
Thank you and sorry for spamming