Xilinx/Vitis_Libraries #57: How to check the result stored in the tab_c

BiscuitsColonel, 2021-03-20:
Hello,

I tested test_q1.exe in the database library.
The following is the execution log.

```
INFO: selected device 0: xilinx_u280_xdma_201920_3
INFO: initilized context.
INFO: initilized command queue.
INFO: created program with binary /Vitis_Libraries/database/L3/build/vt_database_aggr-xilinx_u280_xdma_201920_3.xclbin
INFO: built program.
create gqe::SOL2
sec_l:3, log_part:2, slice_num:4

partition + pipelined aggregate
------------------- Partitioning L table -----------------
=================
aggr_in_nrow_max = 3004998
xxxxxxxxxxxxxxxxxxxxxx invalid below xxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxx invalid above xxxxxxxxxxxxxxxxxxxxxxxx
output nrow[0] = 2
output aggr_sum_nrow[0] = 2
output nrow[1] = 1
output aggr_sum_nrow[1] = 3
output nrow[2] = 1
output aggr_sum_nrow[2] = 4
output nrow[3] = 0
output aggr_sum_nrow[3] = 4
-----------------------Data Transfer Info-----------------------
H2D size = 160.25 MB
D2H size = 0.000305176 MB
------------------------Performance Info------------------------
Tab L pipelined partition, time: 0.116971 ms, throughput: 1337.89 GB/s
aggr pipelined, time: 0.138551 ms, throughput: 1129.5 GB/s
All time: 0.255522 ms, throughput: 612.448 GB/s
Aggr done, table saved: 4 rows,10 cols
```

In the test, I could confirm that the database library was working, but I could not confirm the result of executing the query with q1 of TPC-H.
I think the contents are saved in tab_c, but please tell me how to check.

vt-lib-support, 2021-03-22:
The test code has a command line option `-validate on` to call software reference in `q1.hpp` and compare against FPGA result.

BiscuitsColonel, 2021-03-22:
```
--------------- Query 1, aggregation --------------- 
dat dir is: db_data//dat1
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
file exist, skip generate.
Read Input table form db_data//dat1
Using StrategyManualSet
Select solution:2
Lineitem SF(1)	6001215 rows
Result table	1000 rows
LineItem table has been read from disk
INFO: selected device 0: xilinx_u280_xdma_201920_3
INFO: initilized context.
INFO: initilized command queue.
INFO: created program with binary gqe_aggr_part.xclbin
INFO: built program.
create gqe::SOL2
sec_l:2, log_part:2, slice_num:4

partition + pipelined aggregate
------------------- Partitioning L table -----------------
=================
aggr_in_nrow_max = 3004998
xxxxxxxxxxxxxxxxxxxxxx invalid below xxxxxxxxxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxx invalid above xxxxxxxxxxxxxxxxxxxxxxxx
output nrow[0] = 2
output aggr_sum_nrow[0] = 2
output nrow[1] = 1
output aggr_sum_nrow[1] = 3
output nrow[2] = 1
output aggr_sum_nrow[2] = 4
output nrow[3] = 0
output aggr_sum_nrow[3] = 4
-----------------------Data Transfer Info-----------------------
H2D size = 160.25 MB
D2H size = 0.000305176 MB
------------------------Performance Info------------------------
Tab L pipelined partition, time: 0.155982 ms, throughput: 1003.28 GB/s
aggr pipelined, time: 0.136449 ms, throughput: 1146.9 GB/s
All time: 0.292431 ms, throughput: 535.148 GB/s
Aggr done, table saved: 4 rows,10 cols
Result after OrderBy in CPU:
Q1 Demo Finished!
---------------Validate results---------------
Run Q1 in CPU as Golen
Q1 CPU run done,Start Validate...
Output row number check passed,start checking each value...
0 Errors found.
Validate Pass
```

I'm sorry to say that it's hard to understand.
What you want to check is the contents of tab_c.
Is it possible to check what kind of result is stored in tab_c by referring to [this](https://github.com/Xilinx/Vitis_Libraries/blob/master/database/L3/tests/gqe/aggregate/host/q1.hpp)?

closed 2021-03-23 by BiscuitsColonel