Xilinx/Vitis_Libraries #58: [Security lib] Help on modularExp usage?

lmaxeniro, 2021-03-25:
Hi I am doing some ECC relevant project by using security lib.
I need an Exponentiation Modular function, which mean I need make something like y=x^(e) % m for given big number as below.
`ap_uint<256> x;
ap_uint<256> y;
ap_uint<256> e;
ap_uint<256> m;
`
I find the modularExp in Modular.hpp could be what I need but not 100% sure. the help are simple:
my question:
Q1: Is this the right function that I need? I assume here base=x, exponent = e, modulus =m; 
Q2: if Q1 answer is right,  what is rMod defined? I don't understand the statement of "rMod 2^(2*L) mod modulus could be pre-calculated." --how can I use that parameter?

` * @brief Modular Exponentiation
 *
 * @tparam L bit width of base and modulus
 * @tparam N bit width of exponent
 *
 * @param base Base of Modular Exponentiation
 * @param exponent Exponent of Modular Exponentiation
 * @param modulus Modulus of Modular Exponentiation
 * @param rMod 2^(2*L) mod modulus could be pre-calculated.
 */
template <int L, int N>
ap_uint<L> modularExp(ap_uint<L> base, ap_uint<N> exponent, ap_uint<L> modulus, ap_uint<L> rMod),`

vt-lib-support, 2021-03-25:
Hi @lmaxeniro .
A1: Yes, it is.
A2: rMod = (1 << (2 * L)) % m. In your case, it will be (1 << 512) % m.

https://github.com/Xilinx/Vitis_Libraries/tree/master/security/L1/tests/modular/modularExp is the HLS test project for modularExp.

lmaxeniro, 2021-03-25:
> A2: rMod = (1 << (2 * L)) % m. In your case, it will be (1 << 512) % m.

Thanks a lot, can you please explain a little more, how is the value L (i.e. 512) is determined? which likewise to be something "pre-calculated value"?

vt-lib-support, 2021-03-25:
In modular exponential "y = x^(e) % m",   x y and m will have same bit width which is "L". But L is not necessary same with bit width of e, which is N.  In your case, bit width of x, y and m is 256, so L= 256. rMod = (1 << (2 * L)) % m.

In crypto application like ECC or ECDSA, the modulus is fixed when the ecliptic curve is fixed. Also the bit width of modulus. In such case, rMod is also fixed. So it could be treated as "pre-calculated".

lmaxeniro, 2021-03-25:
Great! Thanks a lot for the help.

closed 2021-03-25 by lmaxeniro