Xilinx/Vitis_Libraries #61: Vitis libraries/database group aggregate filter issue

terminator-888, 2021-04-09:
Using GQE L3 binary for group aggregate.

If std::string filter_str == "" (empty string) the GQEngine will stall for unlimited time and needs to be forced to kill.

Ubuntu 18.04
XRT build version: 2.8.743
Build hash: 77d5484b5c4daa691a7f78235053fb036829b1e9
Build date: 2020-11-16 00:19:11
Git branch: 2020.2

vt-lib-support, 2021-04-19:
Hi,
Thanks for reporting the issue, we have reproduced and internally fixed this issue. Fix will be published shortly. 
Best,

terminator-888, 2021-06-23:
Example:
`input column: TypeInt32 key0,key1,key2,key3,val`
`output: c0=key0,c1=key,c2=key2,c3=key3,s0=sum(val)`
`filter_str = "key3>0"` - work only with this condition

The bug was NOT filxed.
1. If `filter_str == ""` - result always will be 0 rows.
2. If `filter_str = "key0>0"` or `"key1>0"` or `"key2>0"` - STALL the card and process needs to be killed and card must be RESET by XBUTIL
3. `filter_str = "val>1 || val<0"` doesn't work: `Err: not supported filter!`