Xilinx/Vitis_Libraries #62: Vitis libraries/database secondary start attempt failed

terminator-888, 2021-04-09:
Using GQE L3 binary for group aggregate.

First start attempt of gqe:Aggregator bigaggr.aggregate(params1) is successful.
But with secondary start bigaggr.aggregate(params2) it fails with:

create gqe::SOL2
sec_l:2, log_part:2, slice_num:4

partition + pipelined aggregate
XRT build version: 2.8.743
Build hash: 77d5484b5c4daa691a7f78235053fb036829b1e9
Build date: 2020-11-16 00:19:11
Git branch: 2020.2
PID: 22692
UID: 1000
[Fri Apr  9 07:44:58 2021 GMT]
HOST: host123
EXE: /home/user/Projects/test_project/cmake-build-debug/test_project
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc
[XRT] ERROR: std::bad_alloc

vt-lib-support, 2021-04-19:
Hi,
Thanks for reporting the issue, we have reproduced and internally fixed this issue. Fix will be published shortly. 
Best,

vt-lib-support, 2021-04-19:
Hi @terminator-888 , we tried to reproduce your situation by call bigaggr.aggregate() twice in turn but not have the issue you mentioned above. Could you share more details on this so we can reproduce and fix?

terminator-888, 2021-04-19:
> Hi @terminator-888 , we tried to reproduce your situation by call bigaggr.aggregate() twice in turn but not have the issue you mentioned above. Could you share more details on this so we can reproduce and fix?

Here is the code:

unsigned long table_l_nrow = 10000000;
unsigned int solution = 2,
            sec_l = 2,
            slice_num = 4,
            log_part = 2;
    using namespace xf::database;

    gqe::utils::MM pg_mm;
    auto *pg_tab_l_col0 = pg_mm.aligned_alloc<int32_t>(table_l_nrow);    
    auto *pg_tab_l_col1 = pg_mm.aligned_alloc<int32_t>(table_l_nrow);    
    auto *pg_tab_l_col2 = pg_mm.aligned_alloc<int32_t>(table_l_nrow);    
    auto *pg_tab_l_col3 = pg_mm.aligned_alloc<int32_t>(table_l_nrow);    
    auto *pg_tab_l_col4 = pg_mm.aligned_alloc<int32_t>(table_l_nrow);    

// Fill pg_tab_l_col0 and others with values

    gqe::Table pg_tab_l;
    pg_tab_l.addCol("key0", gqe::TypeEnum::TypeInt32, pg_tab_l_col0, table_l_nrow);
    pg_tab_l.addCol("key1", gqe::TypeEnum::TypeInt32, pg_tab_l_col1, table_l_nrow);
    pg_tab_l.addCol("key2", gqe::TypeEnum::TypeInt32, pg_tab_l_col2, table_l_nrow);
    pg_tab_l.addCol("key3", gqe::TypeEnum::TypeInt32, pg_tab_l_col3, table_l_nrow);
    pg_tab_l.addCol("val",  gqe::TypeEnum::TypeInt32, pg_tab_l_col4, table_l_nrow);

    gqe::Table pg_tab_c;
    pg_tab_c.addCol("c0", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c.addCol("c1", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c.addCol("c2", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c.addCol("c3", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c.addCol("s0", gqe::TypeEnum::TypeInt32, table_c_nrow);

    gqe::Aggregator bigaggr1(xclbin_path);

    auto smanual1 = new gqe::AggrStrategyManualSet(solution, sec_l, slice_num, log_part);
    string filter_str1 = "key3>0",    //-------- filter issue, couldn't be ""
            group_keys_str1 = "key0,key1,key2,key3,val",
            ouput_str1 = "c0=key0,c1=key1,c2=key2,c3=key3,s0=sum(val)";

    gqe::ErrCode err_code1;

    err_code1 = bigaggr1.aggregate(pg_tab_l, {{"val", {0}},
                                         {"val", {0}}},
                                 filter_str1, group_keys_str1, ouput_str1, pg_tab_c, smanual1);

    ///----- Secondary execution attempt

    gqe::Table pg_tab_c2;
    pg_tab_c2.addCol("c0", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c2.addCol("c1", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c2.addCol("c2", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c2.addCol("c3", gqe::TypeEnum::TypeInt32, table_c_nrow);
    pg_tab_c2.addCol("s0", gqe::TypeEnum::TypeInt32, table_c_nrow);

    err_code1 = bigaggr1.aggregate(pg_tab_l, {{"val", {0}},
                                              {"val", {0}}},
                                   filter_str1, group_keys_str1, ouput_str1, pg_tab_c2, smanual1);

-----------------
after secondary execution attempt - it fails with errors that I post up here.

terminator-888, 2021-07-22:
Now it works :+1:

closed 2021-07-22 by terminator-888