Xilinx/Vitis_Libraries #64: LZ4 compatibility of data_compress library

BiscuitsColonel, 2021-04-14:
Hello,

I'm currently testing LZ4 acceleration with the Vitis data_compress library.
I was interested in it during that time, so I would like to ask a question.
How can I decompress a file compressed using Vitis's LZ4 compression library with a regular LZ4?
In my verification, the decompression process failed as follows:

```
---Check the test file
# cat test.txt
1, oranges
2, apples
3, bananas
4, strawberries
5, grapes

---Compression process
# ./xil_lz4 -cx compress.xclbin -c test.txt
Found Platform
Platform Name: Xilinx
Found Device=xilinx_u280_xdma_201920_3
INFO: Reading compress.xclbin
Loading: 'compress.xclbin'
E2E(MBps)		:0.02
LZ4_CR			:0.92
File Size(B)		:47.000
File Name		:test.txt

Output Location: test.txt.lz4
Compressed file size: 51

---Check the compressed file
# ls
compress.ltx                    link
compress.xclbin                 logs
compress.xclbin.info            test.txt
compress.xclbin.link_summary    test.txt.lz4
decompress.ltx                  v++_link_compress_guidance.json
decompress.xclbin               v++_link_compress_guidance.pb
decompress.xclbin.info          v++_link_decompress_guidance.json
decompress.xclbin.link_summary  v++_link_decompress_guidance.pb
emconfig.json                   xil_lz4
ext_metadata.json               xsa.xml

---Decompression processing with normal LZ4
# lz4 -d test.txt.lz4
Decoding file test.txt 
test.txt already exists; do you wish to overwrite (y/N) ? y
Error 66 : Decompression error : ERROR_headerChecksum_invalid

---Decompression processing with Vitis LZ4
# ./xil_lz4 -dx decompress.xclbin -d test.txt.lz4 
Found Platform
Platform Name: Xilinx
Found Device=xilinx_u280_xdma_201920_3
INFO: Reading decompress.xclbin
Loading: 'decompress.xclbin'
E2E(MBps)		:0.03
File Size(B)		:70.000
File Name		:test.txt.lz4

Output Location: test.txt.lz4.orig

---Check the test file
# cat test.txt.lz4.orig
1,masayakatoh
2,insight
3,tomato
4,supli
5,fun
```

Does this indicate that LZ4 in the Vitis data_compress library is incompatible with regular LZ4?
Or, it is possible to decompress with normal LZ4, but is the procedure wrong?
please tell me.

kaliudayxilinx, 2021-04-14:
@BiscuitsColonel Thanks for reporting this issue. This is a corner case scenario for which our next public release will provide a fix. Meanwhile please proceed to try large files. We do have compatibility between standard LZ4 & Xilinx LZ4.

BiscuitsColonel, 2021-04-15:
Ok, thanks.

closed 2021-04-15 by BiscuitsColonel