Xilinx/Vitis_Libraries #65: An error occurred while testing the data_analytics library.

BiscuitsColonel, 2021-04-15:
Hi, 

I'm currently testing the data_analytics library.

I tested it using the attached program.
In most cases the test will work, but if I test with the following steps, it will not work.

```
# cat test.log 
pineapple
green apple
apple

# ./test.exe -xclbin reEngineKernel.xclbin -in test.log -out result.log -pt app
----------------------log analytics with regex----------------
INFO: selected device xilinx_u280_xdma_201920_3
INFO: initilized context.
INFO: initilized command queue.
INFO: created program with binary reEngineKernel.xclbin
INFO: built program.
The log file is partition into 1 section with max_slice_lnm 5 and  takes 0.002000 ms.
DEBUG: reEngineKernel has 8 CU(s)
regex pipelined, time: 1.298 ms, size: 3.05176e-05 MB, throughput: 2.29602e-05 GB/s
-----------------------------Finished regex pipelined test----------------------------------------------

ERROR: undefined error code
ERROR: result mismatch

# cat result.log 
Mismatch
0: [-1, -1]
Mismatch
0: [-1, -1]
Match
0: [0, 3]
```

I would appreciate it if you could check the above results and tell me how to solve it.

Thanks.

BiscuitsColonel, 2021-04-15:
```
/*
 * Copyright 2020 Xilinx, Inc.
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include <cstdlib>
#include <fstream>

#include "xf_data_analytics/text/regex_engine.hpp"
#include "general_config.hpp"

// for validating result.
extern "C" {
#include "oniguruma.h"
}

enum {
    MAX_MSG_DEPTH = 250000000,   // Max number of messages in a section
    MAX_MSG_LEN = 65536,         // Max length of message in byte
    MAX_LNM = 6000000,           // Max number of lines in a single section
    MAX_OUT_DEPTH = MAX_LNM * 20 // 20 for 19 capturing groups at most
};
int check_result(std::string pattern,
                 uint64_t* msg_buff,
                 uint32_t* offt_buff,
                 uint16_t* len_buff,
                 uint32_t* out_buff,
                 uint32_t lnm,
                 uint32_t cpgp_nm) {
    int r;
    unsigned char *start, *range, *end;
    regex_t* reg;
    OnigErrorInfo einfo;
    OnigRegion* region = onig_region_new();
    OnigEncoding use_encs[1];

    use_encs[0] = ONIG_ENCODING_ASCII;
    onig_initialize(use_encs, sizeof(use_encs) / sizeof(use_encs[0]));

    UChar* pattern_c = (UChar*)pattern.c_str();

    r = onig_new(&reg, pattern_c, pattern_c + strlen((char*)pattern_c), ONIG_OPTION_DEFAULT, ONIG_ENCODING_ASCII,
                 ONIG_SYNTAX_DEFAULT, &einfo);
    if (r != ONIG_NORMAL) {
        char s[ONIG_MAX_ERROR_MESSAGE_LEN];
        onig_error_code_to_str((UChar*)s, r, &einfo);
        fprintf(stderr, "ERROR: %s\n", s);
        return -1;
    }
    unsigned char* max_str = (unsigned char*)malloc(MAX_MSG_LEN);
    for (int i = 0; i < lnm; ++i) {
        // generate referecne
        int offt = offt_buff[i];
        memcpy(max_str, &msg_buff[offt], len_buff[i]);
        max_str[len_buff[i]] = '\0';
        UChar* str = (UChar*)max_str;
        end = str + strlen((char*)str);
        start = str;
        range = end;
[... 174 lines cut ...]
        }
        if (lnm > 0) {
            // call reInst to do regex
            err_code = reInst.match(lnm, msg_buff, offt_buff, len_buff, out_buff);
            if (err_code) {
                std::cerr << "ERROR: match failed.\n";
                return -1;
            }
            // write data to disk
            store_dat(out_file, out_buff, lnm, cpgp_nm);
            // if check is open, check the result with golden
            int r = check_result(pattern, msg_buff, offt_buff, len_buff, out_buff, lnm, cpgp_nm);
            if (r == -1) {
                fprintf(stderr, "ERROR: result mismatch\n");
            } else {
                fprintf(stdout, "SUCCESS: result match\n");
            }
        }
    }
    log_file.close();
    out_file.close();
    std::quick_exit(0);
    return 0;
}
```

vt-lib-support, 2021-04-19:
Hi,

Thank you for bringing this to our attention. We are implementing the regex match behavior similar to Pytyon's [re.match](https://docs.python.org/3/library/re.html#re.Pattern.match), which _"return a corresponding match object if zero or more characters at the beginning of string match this regular expression"_.

So pattern `app` will match `apple` but will not match `pineapple`.

We will improve documentation to make this more clear in next release.

Thanks!

BiscuitsColonel, 2021-04-19:
I understand the function.
Thanks.

closed 2021-04-19 by BiscuitsColonel