Xilinx/Vitis_Libraries #66: zlib compatibility of data_compression library

LittleWuCoding, 2021-04-16:
Recently I test the zlib you provided in data_compression module.
I find that xilinx's zlib API is not compatible with software version zlib which is running on CPU,that is:
xilinx's API is like:
```
uint64_t xfZlib::compress_file(std::string& inFile_name, std::string& outFile_name, uint64_t input_size, int cu)
```
but software version zlib ilkes:
```
ZEXTERN int ZEXPORT compress OF((Bytef *dest, uLongf *destLen,
                                 const Bytef *source, uLong sourceLen));
```
which they has different  input param. Also, software version zlib can set the compress level from 0 -9.
So, my question is that :
Does  xilinx's zlib API is compatible with software version zlib? that means if I compress data with software version zlib,can I decompress it with zlib API running in FPGA? and the opposite situation?
If the answer is yes, Is there any docs or demo can show it?
If  the answer is no, do you have any plan on the compatibility ?

kaliudayxilinx, 2021-04-19:
@LittleWuCoding We support standard ZLIB compatibility (compress/decompress vice versa). Compress level for FPGA based ZLIB doesnt vary much (Compression Ratio of FPGA ZLIB is around level 1 to level 4 of standard ZLIB) as main focus is on throughput.
compress_file() API must generate ZLIB compatible stream, what did you observe and can you share the demo that you have tried ?

kaliudayxilinx, 2021-04-20:
@LittleWuCoding we have a demo that demonstrates usage of standard ZLIB APIs for FPGA acceleration.
[https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L3/demos/libzso](libzso demo)

Below is link to use of compress2 API which is called internally by compress() API of standard ZLIB.
[https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/data_compression/L3/demos/libzso/src/host.cpp#L123](compress2 API Usage demo)

vt-lib-support, 2021-08-04:
Closing this issue as there is no further activity.

closed 2021-08-04 by vt-lib-support