Xilinx/Vitis_Libraries #69: Cannot synthesize with a FIFO with a depth other than 2

goekce, 2021-04-20:
I want to synthesize [PYNQ-Helloworld](https://github.com/Xilinx/PYNQ-HelloWorld) using the newest Vitis library and get the following error:
```
ERROR: [HLS 214-160] in function 'resize_accel': Local stream 'in_mat.4' has conflict depth setting
```
if I set `depth` to a value other than `2`:
```cpp
#pragma HLS stream variable=in_mat.data depth=3
```

What I did was to take the [L2/resize example](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2/examples/resize) and replace the `m_axi` interface with an `axis` interface:

https://mygit.th-deg.de/gaydos/image-resizer/-/blob/main/xf_resize_accel.cpp

Is this a bug in the new Vitis library or do I use `xf::cv:resize()` incorrectly?

---

To create the error:
```sh
git clone https://mygit.th-deg.de/gaydos/image-resizer
cd image-resizer
make vitis_lib  # clones latest Vitis-Libraries
# change the depth from 2 to 3 in xf_resize_accel.cpp
make hls
```

vt-lib-support, 2021-04-21:
Hi @goekce ,

`#pragma HLS stream` is no longer used to set the depth. This pragma usage will be removed from the code in next release.

If you would like to modify the existing depth value, please specify the new value in the xf::cv::Mat declaration in the [accel ](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L2/examples/resize/xf_resize_accel.cpp#L36) file and comment out the `#pragma HLS stream` [line](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L2/examples/resize/xf_resize_accel.cpp#L38).

**Example** :    ` xf::cv::Mat<TYPE, HEIGHT, WIDTH, NPC_T, 3> in_mat(rows_in, cols_in);`

Here, the last template parameter '3' is the user specified depth value. If you don't specify anything, the stream depth will be set to the default value of 2. 
See the [xf::cv::Mat](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L1/include/common/xf_structs.hpp#L427) definition for more info.

goekce, 2021-04-21:
:heart: for your fast response!

Now I amended my code:
```cpp
xf::cv::Mat<TYPE, HEIGHT, WIDTH, NPC_T, 150> in_mat(rows_in, cols_in);  
xf::cv::Mat<TYPE, NEWHEIGHT, NEWWIDTH, NPC_T, 150> out_mat(rows_out, col  s_out);                                                                     
                                                                                                                                                        
xf::cv::axiStrm2xfMat<INPUT_PTR_WIDTH, TYPE, HEIGHT, WIDTH, NPC_T>(
    img_inp, in_mat);
xf::cv::resize<INTERPOLATION, TYPE, HEIGHT, WIDTH, NEWHEIGHT, NEWWIDTH,   NPC_T, MAXDOWNSCALE>(
    in_mat, out_mat);
xf::cv::xfMat2axiStrm<OUTPUT_PTR_WIDTH, TYPE, NEWHEIGHT, NEWWIDTH, NPC_T  >(
    out_mat, img_out);
```
and I get:
```
ERROR: [HLS 207-3334] no matching function for call to 'resize': xf_resize_accel.cpp:28:5
INFO: [HLS 207-4373] candidate function not viable: no known conversion from 'xf::cv::Mat<XF_8UC3, 1080, 1920, XF_NPPC1, 150>' to 'xf::cv::Mat<9, 1080, 1920, 1> &' for 1st argument: vitis_lib/vision/L1/include/common/xf_utility.hpp:534:6
```

Does `xf::cv::resize` support an `xfMat` with depth?:

https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L1/include/imgproc/xf_resize.hpp#L28-L36

goekce, 2021-04-21:
Actually before `xf::cv::resize()` the utility function `axiStrm2xfMat()` errors out:
```
ERROR: [HLS 207-3334] no matching function for call to 'axiStrm2xfMat': xf_resize_accel.cpp:27:
INFO: [HLS 207-4373] candidate function not viable: no known conversion from 'xf::cv::Mat<XF_8UC3, 2160, 3840, XF_NPPC1, 150>' to 'xf::cv::Mat<9, 2160, 3840, 1> &' for 2nd argument: vitis_lib/vision/L1/include/common/xf_utility.hpp:540:6
```
looks like `axiStrm2xfMat` also has the same problem as `xf::cv::resize()`

vt-lib-support, 2021-04-21:
If you are using `hls::stream<ap_axiu ...... >` at the interface, [these ](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L1/examples/axiconv/xf_axiconv_accel.cpp#L33) are the recommended converter functions to use. 

Even these functions need to be modified to make them work for custom depths.

[AXIvideo2xfMat](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L1/include/common/xf_infra.hpp#L99-L100) :
```
template <int W, int TYPE, int ROWS, int COLS, int NPPC>
int AXIvideo2xfMat(hls::stream<ap_axiu<W, 1, 1, 1> >& AXI_video_strm, xf::cv::Mat<TYPE, ROWS, COLS, NPPC>& img) 
```
In the above function, add an extra template parameter for depth and use it in xf::cv::Mat argument, as shown below:

```
template <int W, int TYPE, int ROWS, int COLS, int NPPC, int __XFCV_DEPTH>
int AXIvideo2xfMat(hls::stream<ap_axiu<W, 1, 1, 1> >& AXI_video_strm, xf::cv::Mat<TYPE, ROWS, COLS, NPPC, __XFCV_DEPTH>& img)
```

Similarly the [xfMat2AXIvideo ](https://github.com/Xilinx/Vitis_Libraries/blob/eb47ae535f1476c967ffb90a55799f8d97e4d9d4/vision/L1/include/common/xf_infra.hpp#L178-L179) function needs to be modified.

goekce, 2021-04-21:
it is a deep rabbit hole.

`xf::cv::resize()` template must be modified, which calls many other Vision functions. I stopped at:

https://github.com/Xilinx/Vitis_Libraries/blob/4a66c3276d066a9f49751a4df52e8c4f5d9e00e6/vision/L1/include/imgproc/xf_resize_up_area.hpp#L161-L171

My impression is that most of the Vitis library is not depth aware. Is this a bug or a feature?

Maybe that was the reason why `PYNQ-Helloworld` does not use any of the Vitis converter functions but defines its own:

https://github.com/Xilinx/PYNQ-HelloWorld/blob/b9ec77a8d01f8b77acb9a1ca5047b3a0866c6e49/boards/ip/hls/resize/xf_resize_accel.cpp#L33-L65

vt-lib-support, 2021-08-17:
Vitis Vision functions by default does not support providing custom depths to its functions from the user. 
If it is required for a given function, all the xf::cv::Mat instances needs to be updated to the desired depth value, by specifying the extra template parameter in the Mat declaration(s). Similarly for any instances of hls::streams, required depth should be updated manually by the user.

closed 2021-08-17 by vt-lib-support