Xilinx/Vitis_Libraries #75: Vitis VISON：ERROR when compile host.exe

xiaoliang201188, 2021-07-27:
do you have any idea ?
[image]

giorgiob-xilinx, 2021-07-27:
All tests have been upgraded from using OpenCV 3.4.2 to OpenCV 4.4.0 

You need to [download](https://opencv.org/opencv-4-4-0/) and build it, and pass to the makefile the variables
OPENCV_INCLUDE
OPENCV_LIB
accordingly to your OpenCV installations.

vt-lib-support, 2021-07-27:
@xiaoliang201188 
Also, please make sure to set the LD_LIBRARY_PATH to point to your OpenCV libs as mentioned in the [L1 README](https://gitenterprise.xilinx.com/FaaSApps/xf_opencv/blob/next/L1/README.md)

xiaoliang201188, 2021-07-28:
> All tests have been upgraded from using OpenCV 3.4.2 to OpenCV 4.4.0 ([README.md](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/README.md))
> 
> You need to [download](https://opencv.org/opencv-4-4-0/), build it, and pass to the makefile the variables
> OPENCV_INCLUDE
> OPENCV_LIB
> accordingly to your OpenCV installations.

Hi，I use tag19.2，is it must update the opencv ?

xiaoliang201188, 2021-07-28:
> @xiaoliang201188
> Also, please make sure to set the LD_LIBRARY_PATH to point to your OpenCV libs as mentioned in the [L1 README](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L1/README.md)

Hi，I have set :
     export OPENCV_INCLUDE=/usr/local/include/opencv2
     export OPENCV_LIB=/usr/local/lib64/
     export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib64
It's invalid...