Xilinx/Vitis_Libraries #76: Vitis_Libraries/data_compression/L2/tests/gzipc/description.json

wanyuzhan, 2021-08-01:
Hi, in order to run gzipc in Vitis GUI, I tried to find all kernel codes in the gzipc project. Shall I know which files belong to the kernel?
Also, some files not found in kernel when I did the software emulation.  Shall I know how to solve it?
[image]

meghana-t, 2021-09-01:
Hi @wanyuzhan 
1. You can trace kernel files from description.json "accelerator" section: 
[image]

For L2/tests/gzipc project, kernel files are found at https://github.com/Xilinx/Vitis_Libraries/blob/master/data_compression/L2/src/gzip_compress_multicore_stream.cpp as mentioned in description.json. 

2. All the kernel include files are added in description.json as part of vitis " v++ compiler / includepaths". 
[image]
https://github.com/Xilinx/Vitis_Libraries/blob/master/data_compression/L2/tests/gzipc/description.json#L66

3. Could you please create application project in Vitis GUI as and try software emulation again ? I don't find any kernel files missing in this case. 
[image]

Let me know if you still face any other issues.

wanyuzhan, 2021-09-01:
Thanks for your reply. I downloaded Vitis_Libraries in Vitis GUI and ran Vitis_Libraries/data_compression/L2/tests/gzipc. In this case, I successfully did the software emulation. However, when I launched the software,  I encountered the following error. I sincerely hope to get your help。
[image]

meghana-t, 2021-09-02:
Hi @wanyuzhan 
I hope you are trying correct steps while building for software emulation first. You can check below steps and follow the same. 

1. create application project 
[image]

2. Build for complete **system project** 
[image]

3. Once build is successful, xclbin is generated. 
[image]

4. Launch / run software emulation using [image]

5. Check for console output on successful run.

wanyuzhan, 2021-09-02:
Thanks for your tips. I need to say I run the program using the steps you mentioned. But when I Launch software emulation, I met this error. 
[image]

meghana-t, 2021-09-02:
Hi @wanyuzhan 
Can you please point me to the gzip_system_hw_link directory in the explorer view ? Need to make sure if xclbin is actually generated or its unreadable. 
Also, make sure you setting up the vitis environment correctly https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/settingupvitisenvironment.html#zks1565446519267 before launching the vitis IDE. For complete GUI flow, please refer below: 
https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/creatingvitisideproject.html#cpe1508968036414

meghana-t, 2021-09-02:
Hi @wanyuzhan 
Could you please share the complete steps you tried alongwith the environment settings ?

wanyuzhan, 2021-09-07:
Thanks for your reply. Now I launch this program successfully and get this following result. However,I can't find a file from which I can get this result. Also, Which is the file that reads the sample.txt?
[image]

meghana-t, 2021-09-07:
Hi @wanyuzhan 
You can check the input <sample.txt> and generated output <sample.txt.gz> in data folder as below: 
[image]

Same location you can trace using terminal (CLI) from your workspace area as : /<path/to/workspace>/gzip/data/

wanyuzhan, 2021-09-07:
Thanks for your reply! I'm sorry to have failed to make myself clear. My question is which part of the code in this project is used to read input file(sample.txt)? I want to read a different file instead of sample.txt but I didn't find where to change.

meghana-t, 2021-09-07:
Hi @wanyuzhan 
You can update arguments in run configuration to parse new file, please follow below steps: 
1. Click on run configuration 
[image]

2. Click Edit .. next to program arguments
[image]

3. Click to edit and update the new filename. Make sure new file is present at correct location. 
[image]

4. Proceed to click OK .. and click Run. 
[image]

5. Check the console output for successful run. 

File read / handling is done here: https://github.com/Xilinx/Vitis_Libraries/blob/master/data_compression/common/libs/compress/compressApp.cpp#L230

wanyuzhan, 2021-09-07:
Thanks for your help! I already know IDE usage and I believe this is a good start. I was grateful for your wise counsel.

heeran-xilinx, 2021-09-07:
Thanks @wanyuzhan . We  are closing this issue assuming that it is resolved.

closed 2021-09-07 by heeran-xilinx