Xilinx/Vitis_Libraries #77: CSim failure while buidling L1 examples

kinaikan, 2021-08-05:
Hi,
I'm trying to build one of the L1 examples ( `resize` ), targeting `xc7z020-clg400-1`, however no matter what settings I try, C Simulation always fails with following error:

[image]

I'm trying to build the project on my x64 Windows 10 `build 19042.1110` machine. I'm using Vitis HLS 2020.2 `build 3064766`.
I compiled and build opencv-3.4.11 CMake and MinGW using MinGW configurations in CMake.

my environmental variables are set as noted in Vitis documents:

[image]

I tried to build the project using .tcl provided [here](https://www.xilinx.com/support/answers/75727.html) and also using GUI.

in both cases I also setup the Linker Flags and CFlags.
 
but I always end up here:

[image]
with this particular example, I just did a slightly modification to CFlag values and changed the `-std=c++0x` to `-std=c++14`.
 cause while I try to build with `-std=c++0x` I get some warnings about template files which are just supported  with c++14.

Following the trace logs, doesn't tell anything particular to me (sorry if they have to; it's my bad).

any help or suggestion would be appreciated, cause I don't have any clue what I'm doing wrong.

Thanks

vt-lib-support, 2021-08-17:
Hi @kinaikan,

Vitis Vision Libraries are not officially supported on Windows OS. 

We suggest you to use any of the supported operating systems listed [here](https://www.xilinx.com/html_docs/xilinx2021_1/vitis_doc/acceleration_installation.html#ariaid-title2).

closed 2021-08-17 by vt-lib-support

m-tosch, 2022-07-15:
Is there a plan to support Windows OS for the Vitis Vision Libraries?