Xilinx/Vitis_Libraries #78: Documentation is messed up an not responsive

vmayoral, 2021-08-25:
While browsing through the documentation, the current template being used does not expand options on the left bar column appropriately (when the options are more than the height of the screen). See the following picture:

[image]

I can only see Function until `BoundingBox` clearly. Anything below that, it's hidden due the non-responsiveness of the template. This happens across other Xilinx documents.

vt-lib-support, 2021-08-25:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

@Brydon J ***@***.***>

Can you take a look at this and see if you can address the documentation issues?

Regards,
Chris

[imagen]<https://user-images.githubusercontent.com/1375246/130769389-78912d0b-aa6b-4492-91d8-1f9dd0ec0148.png>

vt-lib-support, 2021-11-10:
Scrolling issue fixed in the [latest release](https://xilinx.github.io/Vitis_Libraries/vision/2021.2/index.html).

closed 2021-11-10 by vt-lib-support