Xilinx/Vitis_Libraries #79: Can't build a simple vision/L1 example

vmayoral, 2021-08-29:
I'm trying to build a simple Vitis Vision Library L1 example using Vitis `2020.2`. 
Here's my setup:

```bash
$ pwd
/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize
```

Then build following [instructions](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L1#commands-to-run):

```bash
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm
export OPENCV_INCLUDE=/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/include
export OPENCV_LIB=/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/lib/libopencv_imgproc.so
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/lib
make run CSIM=1 CSYNTH=0 COSIM=0
```

This results into a number of failures which are available below:

<details><summary>errors</summary>

```bash
xilinx@xilinx:~/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize$ make run CSIM=1 CSYNTH=0 COSIM=0
Configured: settings.tcl
----
set XPART XCK26-SFVC784-2LV-C
set CSIM 1
set CSYNTH 0
set COSIM 0
set VIVADO_SYN 0
set VIVADO_IMPL 0
set XF_PROJ_ROOT "/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/"
set OPENCV_INCLUDE "/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/include"
set OPENCV_LIB "/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/lib/libopencv_imgproc.so"
set CUR_DIR "/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize"
----
vitis_hls -f run_hls.tcl;

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2020.2.2 (64-bit)
  **** SW Build 3118627 on Tue Feb  9 05:13:49 MST 2021
  **** IP Build 3115676 on Tue Feb  9 10:48:11 MST 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vitis_HLS/2020.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vitis_HLS/2020.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'xilinx' on host 'xilinx' (Linux_x86_64 version 5.11.0-27-generic) on Sun Aug 29 12:24:03 CEST 2021
INFO: [HLS 200-10] On os Ubuntu 20.04.2 LTS
INFO: [HLS 200-10] In directory '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize'
Sourcing Tcl script 'run_hls.tcl'
INFO: [HLS 200-1510] Running: open_project -reset resize.prj
INFO: [HLS 200-10] Opening and resetting project '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj'.
WARNING: [HLS 200-40] No /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1510] Running: add_files /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_accel.cpp -cflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x
INFO: [HLS 200-10] Adding design file '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_tb.cpp -cflags -I/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/include -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x
INFO: [HLS 200-10] Adding test bench file '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_tb.cpp' to the project
INFO: [HLS 200-1510] Running: set_top resize_accel
INFO: [HLS 200-1510] Running: open_solution -reset sol1
INFO: [HLS 200-10] Creating and opening solution '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1'.
INFO: [HLS 200-10] Cleaning up the solution database.
WARNING: [HLS 200-40] No /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1505] Using default flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2020.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part XCK26-SFVC784-2LV-C
INFO: [HLS 200-10] Setting target device to 'xck26-sfvc784-2LV-c'
INFO: [HLS 200-1510] Running: create_clock -period 3.3
INFO: [SYN 201-201] Setting up clock 'default' with a period of 3.3ns.
INFO: [HLS 200-1510] Running: csim_design -ldflags -L /home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/lib/libopencv_imgproc.so -lopencv_imgcodecs -lopencv_imgproc -lopencv_core -lopencv_highgui -lopencv_flann -lopencv_features2d -argv  /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//data/128x128.png
[... 2076 lines cut ...]
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:835:36: error: ‘__gthread_recursive_mutex_t’ was not declared in this scope
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:835:65: error: ‘__mutex’ was not declared in this scope
 __gthread_recursive_mutex_destroy (__gthread_recursive_mutex_t *__mutex)
                                                                 ^~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:850:27: error: ‘__gthread_cond_broadcast’ declared as an ‘inline’ variable
 __gthread_cond_broadcast (__gthread_cond_t *__cond)
                           ^~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:850:27: error: ‘__gthread_cond_t’ was not declared in this scope
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:850:45: error: ‘__cond’ was not declared in this scope
 __gthread_cond_broadcast (__gthread_cond_t *__cond)
                                             ^~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:856:24: error: ‘__gthread_cond_signal’ declared as an ‘inline’ variable
 __gthread_cond_signal (__gthread_cond_t *__cond)
                        ^~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:856:24: error: ‘__gthread_cond_t’ was not declared in this scope
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:856:42: error: ‘__cond’ was not declared in this scope
 __gthread_cond_signal (__gthread_cond_t *__cond)
                                          ^~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:862:22: error: ‘__gthread_cond_wait’ declared as an ‘inline’ variable
 __gthread_cond_wait (__gthread_cond_t *__cond, __gthread_mutex_t *__mutex)
                      ^~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:862:22: error: ‘__gthread_cond_t’ was not declared in this scope
/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/include/c++/6.2.0/x86_64-pc-linux-gnu/bits/gthr-default.h:862:40: error: ‘__cond’ was not declared in this scope
 __gthread_cond_wait (__gthrea

vt-lib-support, 2021-08-30:
Hi @vmayoral ,

I'm assuming you are building this example on a x86 CPU. If yes, the L1 examples expects OpenCV _includes_ and _libs_ also for x86. Hence the OPENCV_INCLUDE and OPENCV_LIB variables should point to the x86 version and not the aarch64 libs that you seem to be using from the sysroot path. 
Please do this change and let know if you still see any issues.

vmayoral, 2021-08-30:
@vt-lib-support thanks for the quick reaction. Appreciate it.

As a side comment:

> I'm assuming you are building this example on a x86 CPU. If yes, the L1 examples expects OpenCV includes and libs also for x86.

Then this should be specified clearly, and an example should be provided (even if hardcoded for a particular setup) so that users have a good intuition of what's expected. There're too many assumption of previous background and familiarity with tools across the libraries. Note there's prior evidence showing how the current setup leads to confusion (e.g. [here](https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/Vitis-vison-linker-errors/td-p/1158108)).

I did what's suggested:

```
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm
export OPENCV_INCLUDE=/usr/include/opencv4
export OPENCV_LIB=/usr/lib/x86_64-linux-gnu/
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/x86_64-linux-gnu/
make run CSIM=1 CSYNTH=0 COSIM=0
```

and still crashes:

<details><summary>dump</summary>

```bash
xilinx@xilinx:~/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize$ make run CSIM=1 CSYNTH=0 COSIM=0
Configured: settings.tcl
----
set XPART XCK26-SFVC784-2LV-C
set CSIM 1
set CSYNTH 0
set COSIM 0
set VIVADO_SYN 0
set VIVADO_IMPL 0
set XF_PROJ_ROOT "/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/"
set OPENCV_INCLUDE "/usr/include/opencv4"
set OPENCV_LIB "/usr/lib/x86_64-linux-gnu/"
set CUR_DIR "/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize"
----
vitis_hls -f run_hls.tcl;

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2020.2.2 (64-bit)
  **** SW Build 3118627 on Tue Feb  9 05:13:49 MST 2021
  **** IP Build 3115676 on Tue Feb  9 10:48:11 MST 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vitis_HLS/2020.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vitis_HLS/2020.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'xilinx' on host 'xilinx' (Linux_x86_64 version 5.11.0-27-generic) on Mon Aug 30 10:08:49 CEST 2021
INFO: [HLS 200-10] On os Ubuntu 20.04.2 LTS
INFO: [HLS 200-10] In directory '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize'
Sourcing Tcl script 'run_hls.tcl'
INFO: [HLS 200-1510] Running: open_project -reset resize.prj
INFO: [HLS 200-10] Opening and resetting project '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj'.
WARNING: [HLS 200-40] No /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1510] Running: add_files /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_accel.cpp -cflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x
INFO: [HLS 200-10] Adding design file '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_tb.cpp -cflags -I/usr/include/opencv4 -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/include -I /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/build -I ./ -D__SDSVHLS__ -std=c++0x
INFO: [HLS 200-10] Adding test bench file '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision//L1/examples/resize/xf_resize_tb.cpp' to the project
INFO: [HLS 200-1510] Running: set_top resize_accel
INFO: [HLS 200-1510] Running: open_solution -reset sol1
INFO: [HLS 200-10] Creating and opening solution '/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1'.
INFO: [HLS 200-10] Cleaning up the solution database.
WARNING: [HLS 200-40] No /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/examples/resize/resize.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1505] Using default flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2020.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part XCK26-SFVC784-2LV-C
INFO: [HLS 200-10] Setting target device to 'xck26-sfvc784-2LV-c'
INFO: [HLS 200-1510] Running: create_clock -period 3.3
INFO: [SYN 201-201] Setting up clock 'default' with a period of 3.3ns.
INFO: [HLS 200-1510] Running: csim_design -ldflags -L /usr/lib/x86_64-linux-gnu/ -lopencv_imgcodecs -lopencv_imgproc -lopencv_core -lopencv_highgui -lopencv_flann -lopencv_features2d -argv  /home/xilinx/ros2_ws/src/Vitis_Libraries/vision//data/128x128.png
[... 152 lines cut ...]
/usr/lib/x86_64-linux-gnu//libgeos-3.8.0.so: undefined reference to `std::runtime_error::runtime_error(std::runtime_error&&)@GLIBCXX_3.4.26'
/usr/lib/x86_64-linux-gnu//libopencv_imgcodecs.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
[... cut ...]
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 8.22 seconds. CPU system time: 0.52 seconds. Elapsed time: 8.49 seconds; current allocated memory: 195.020 MB.
4
    while executing
"source run_hls.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vitis_hls at Mon Aug 30 10:30:45 2021...
make: *** [Makefile:249: runhls] Error 1
```

</details>

@vt-lib-support, is there a public CI that shows that this is maintained and *works* as is for `2020.2`?

vt-lib-support, 2021-08-30:
Hi @vmayoral ,

The info related to the OpenCV lib version according to the target OS is mentioned in the [Prerequisites](https://pages.gitenterprise.xilinx.com/FaaSApps/xf_opencv/2020.2/overview.html#prerequisites) section of the documentation. We will enhance this information and also will copy into the README too for easy visibility to the users.

2020.2 Vision library is tested against OpenCV version 3.4.2, as mentioned in the [documentation](https://pages.gitenterprise.xilinx.com/FaaSApps/xf_opencv/2020.2/overview.html#prerequisites). So, the fix that you provided in #81 might not be applicable to 2020.2

The [2021.1](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision) release ( which happens to be the latest release) uses OpenCV 4.4 (mentioned in the [documentation](https://pages.gitenterprise.xilinx.com/FaaSApps/xf_opencv/2021.1/overview.html#prerequisites)) and already uses the APIs that you provided in #81

Looking at the linking errors that you posted, seems like it is error related to incompatibility between the OpenCV libs and the tool's GCC compiler version. Please try building the OpenCV libs compatible to the GCC version that you are using for the CSIM build.

Please let us know for any further assistance.

vmayoral, 2021-08-30:
> 2020.2 Vision library is tested against OpenCV version 3.4.2, as mentioned in the [documentation](https://pages.gitenterprise.xilinx.com/FaaSApps/xf_opencv/2020.2/overview.html#prerequisites). So, the fix that you provided in #81 might not be applicable to 2020.2

I think you meant https://xilinx.github.io/Vitis_Libraries/vision/2020.2/overview.html#getting-started-with-vitis-vision, right?

> Looking at the linking errors that you posted, seems like it is error related to incompatibility between the OpenCV libs and the tool's GCC compiler version. Please try building the OpenCV libs compatible to the GCC version that you are using for the CSIM build.

Thanks for the heads up. Can you be more specific? Which version of the library with which version of the compiler?

prkrmx, 2021-10-01:
hi @vmayoral 
Did you solve it? I have similar errors. In my case i used opencv version 4.4.0 #85