Xilinx/Vitis_Libraries #80: Can't build a simple vision/L2 example

vmayoral, 2021-08-29:
Trying now quickly the examples aimed for Vitis (instead of for HLS which fail, refer to https://github.com/Xilinx/Vitis_Libraries/issues/79) described in [the instructions](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2#commands-to-run).

Using in particular [resize](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2/examples/resize):

```bash
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm
export SYSROOT=/home/xilinx/ros2_ws/install/../acceleration/firmware/kv260/sysroots/aarch64-xilinx-linux
export EDGE_COMMON_SW=/home/xilinx/ros2_ws/acceleration/firmware/select
#export PERL=<path-to-perl-installation-location> #For example, "export PERL=/usr/bin/perl".

make host xclbin TARGET=sw_emu HOST_ARCH=aarch64

Makefile:74: [WARNING]: This project has not been tested for /home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm. It may or may not work.
mkdir -p build_dir.sw_emu.kv260_base
/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++ -o build_dir.sw_emu.kv260_base/resize /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/xf_resize_tb.cpp /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.cpp -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/build -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/ -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/include -D SW_EMU_TEST -I/opt/xilinx/xrt/include -I/tools/Xilinx/Vivado/2020.2/include -std=c++11 -O3 -Wall -Wno-unknown-pragmas -Wno-unused-label -fmessage-length=0 -O3 -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/src/  -lopencv_videoio -lopencv_imgcodecs -lopencv_core -lopencv_imgproc -lopencv_features2d -lopencv_flann -lopencv_video -lopencv_calib3d -lopencv_highgui -L/opt/xilinx/xrt/lib -lOpenCL -lpthread -lrt -Wno-unused-label -Wno-narrowing -DVERBOSE
In file included from /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/xf_resize_tb.cpp:17:
/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/include/common/xf_headers.hpp:23:10: fatal error: opencv2/opencv.hpp: No such file or directory
   23 | #include "opencv2/opencv.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.cpp:33:
/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.hpp:45:10: fatal error: CL/cl2.hpp: No such file or directory
   45 | #include <CL/cl2.hpp> //"/opt/intel/opencl-1.2-4.4.0.117/include/CL/cl.h"
      |          ^~~~~~~~~~~~
compilation terminated.
```

vmayoral, 2021-08-29:
These first few hurdles, in my case, can be bypassed with:

```bash
CXXFLAGS += -I/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/include
LDFLAGS += -L/home/xilinx/ros2_ws/acceleration/firmware/select/sysroots/aarch64-xilinx-linux/usr/lib 
```

Which leads to further issues, so I'm stopping here and asking for support. AFAIK this should work out of the box, but it doesn't. Am I wrong somewhere? Otherwise, please clarify how to get this right.

vmayoral, 2021-08-29:
> make host xclbin TARGET=sw_emu HOST_ARCH=aarch64

For the record, I tried other targets, this is not a target's issue.

vt-lib-support, 2021-08-30:
Hi @vmayoral ,

Few of the platforms specific flags as in [here](https://github.com/Xilinx/Vitis_Libraries/blob/2323dc5036041e18242718287aee4ce66ba071ef/vision/L2/examples/resize/Makefile#L142-L169) in all the Makefiles are set based on the platform.
And, the current format of the Makefiles will have these flags set only for the platforms that are listed in the "platform_whitelist" section of the description.json file, as those platforms only have been tested to work. (A warning appears mentioning that it may or may not work for other platforms other than what were tested, as you can see in your posted log)

Hence if the build is targeted for a different platform, as in your case, these flags need to be manually set in the Makefile and the build is expected to work fine. Please try this and let us know (attaching the log) if issue still persists.

vt-lib-support, 2021-08-30:
This issue is also reported here #43 and we are working on removing this limitation from the Makefiles.

vmayoral, 2021-08-30:
> Few of the platforms specific flags as in [here](https://github.com/Xilinx/Vitis_Libraries/blob/2323dc5036041e18242718287aee4ce66ba071ef/vision/L2/examples/resize/Makefile#L142-L169) in all the Makefiles are set based on the platform.

@vt-lib-support, this points to `2021.1` release. As pointed out, this is connected to https://github.com/Xilinx/Vitis_Libraries/issues/79 which targets `2020.2` (expecting it to be more mature and stable).

Nevertheless, here's my attempt targeting the ZCU102 (which is listed as supported) **on a plain new terminal**:

```bash
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/xilinx/ros2_ws/acceleration/firmware/zcu102/platform/xilinx_zcu102_base_202020_1.xpfm
export SYSROOT=/home/xilinx/ros2_ws/acceleration/firmware/zcu102/sysroots/aarch64-xilinx-linux
export EDGE_COMMON_SW=/home/xilinx/ros2_ws/acceleration/firmware/zcu102

make host xclbin TARGET=sw_emu HOST_ARCH=aarch64
```

<details>

```bash
xilinx@xilinx:~/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize$ make host xclbin TARGET=sw_emu HOST_ARCH=aarch64
Cannot find OpenCV include path. Please set OPENCV_INCLUDE variable
make: *** [Makefile:216: check_env] Error 1
```
</details>

This leads to instructions missing in the README @vt-lib-support. 

I'll try and get a complete setup running by setting the embedded variables right next.

vmayoral, 2021-08-30:
After a while on this, I gave up. There're simply too many things that need to be modified for this to be a useful blueprint at `2020.2`, including a merge with the new makefile logic provided in future releases as hinted in [here](https://github.com/Xilinx/Vitis_Libraries/issues/80#issuecomment-908106207).

I unfortunately don't have more time to help with this but I encourage a huge re-consideration of the instructions provided and build logic (Makefiles essentially). 

After various modifications in the source code, and in the build system provided (I essentially discarded Makefiles and built a complete new CMakeLists.txt file that allowed me to integrate `resize` into other projects I managed to build), **I managed to build things successfully, including the kernel**. I will probably disclose this source code at a later stage, in a different repository (it needs further work), but ping me if anyone needs access to this immediately.

On a side note, @vt-lib-support please consider the input in here. As is, with `2020.2`, the content in here has very limited usefulness, and I doubt users will be willing to fight and struggle a weekend like me just to get a hello world (vision) demo running. Also, I'd highly encourage to set up a public facing CI system that clears all possible doubt. Meaning, complaints like this could be referred to a CI build log which systematically shows evidence that things build indeed with the proper setup. Even better, I'd love to see such builds *dockerized*, so I don't need to spend hours ensuring my system is *"as expected"* by whoever designed this.

The need for dockerized systematic builds across Xilinx's organization's examples is a huge need identified that I very much encourage.

kapoor7997, 2022-02-12:
Hi,

I am facing the same issue with exact hardware zcu102 and libraries.
Can you please provide solution if you have one.

Thanks,
kapoor7997

vt-lib-support, 2022-02-14:
Hi @kapoor7997 ,

Please post your error and the steps you used for the build.

kapoor7997, 2022-02-14:
I followed the same steps as [vmayoral](https://github.com/vmayoral) and got the same error. The error I get is
steps:
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm
export SYSROOT=/home/xilinx/ros2_ws/install/../acceleration/firmware/kv260/sysroots/aarch64-xilinx-linux
export EDGE_COMMON_SW=/home/xilinx/ros2_ws/acceleration/firmware/select
export PERL=<path-to-perl-installation-location> #For example, "export PERL=/usr/bin/perl".
make host xclbin DEVICE=zcu102 TARGET=hw HOST_ARCH=aarch64

Error:
Makefile:74: [WARNING]: This project has not been tested for /home/xilinx/ros2_ws/acceleration/firmware/select/platform/kv260_base.xpfm. It may or may not work.
mkdir -p build_dir.sw_emu.kv260_base
/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++ -o build_dir.sw_emu.kv260_base/resize /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/xf_resize_tb.cpp /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.cpp -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/build -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/ -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/include -D SW_EMU_TEST -I/opt/xilinx/xrt/include -I/tools/Xilinx/Vivado/2020.2/include -std=c++11 -O3 -Wall -Wno-unknown-pragmas -Wno-unused-label -fmessage-length=0 -O3 -I/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/src/  -lopencv_videoio -lopencv_imgcodecs -lopencv_core -lopencv_imgproc -lopencv_features2d -lopencv_flann -lopencv_video -lopencv_calib3d -lopencv_highgui -L/opt/xilinx/xrt/lib -lOpenCL -lpthread -lrt -Wno-unused-label -Wno-narrowing -DVERBOSE
In file included from /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L2/examples/resize/xf_resize_tb.cpp:17:
/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/L1/include/common/xf_headers.hpp:23:10: fatal error: opencv2/opencv.hpp: No such file or directory
   23 | #include "opencv2/opencv.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.cpp:33:
/home/xilinx/ros2_ws/src/Vitis_Libraries/vision/ext/xcl2/xcl2.hpp:45:10: fatal error: CL/cl2.hpp: No such file or directory
   45 | #include <CL/cl2.hpp> //"/opt/intel/opencl-1.2-4.4.0.117/include/CL/cl.h"
      |          ^~~~~~~~~~~~
compilation terminated.

Plese do the needful on that, I realaly need support from the community. IT would be really appriciated if you can provide exact steps to create images and how I can use vitis vision library L2 demo om zcu102 board.

Thanks, 
kapoor7997

vt-lib-support, 2022-02-15:
Hi @kapoor7997 ,

We see that you intend to build the function for zcu102 but have set the DEVICE to kv260 which looks incorrect.

Please follow the steps mentioned in the **Embedded Devices** section of this [README](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L2/README.md#commands-to-run-pl-tests) on how to get the required platform and follow the steps mentioned there to install _sysroot_.

vt-lib-support, 2022-02-16:
Hi @kapoor7997 ,

Since you are already setting the DEVICE as env variable pointing to your platform xpfm file, there is no need to set it again with the make command. So your build command should be:
`make xclbin host TARGET=hw HOST_ARCH=aarch64`

Also, as your platform seems to be custom built (not among the one in _platform_whitelist_ of description.json), please add the following flags in the Makefile:
```
CXXFLAGS += -I$(SYSROOT)/usr/include
CXXFLAGS += --sysroot=$(SYSROOT)
LDFLAGS += -L$(SYSROOT)/usr/lib
LDFLAGS += -L${SYSROOT}/opt/xilinx/xrt/lib
```

kapoor7997, 2022-02-17:
[vitis_hls.log](https://github.com/Xilinx/Vitis_Libraries/files/8090574/vitis_hls.log)
Hi [vt-lib-support](https://github.com/vt-lib-support),

First of I all I want to provide info about my system.
1. I have Vivado, Vitis and Petalinux 2020.2 installed on my ubuntu pc.
2. I am using **trial version** of Xilinx Software and the trial of Vivado and Vitis. **My trial has Expired**. But because I am not creating my own xsa, petalinux should still work for me. Thats what I am assuming.
3. I created the **sysroot** from my **own petalinux project** that I **created from the xsa file provided by xilinx in their Xilinx base image for ZCU102**.
**Does not having a trial license causes petalinux to fail building.?** We are a startup and are looking to do image processing demo to get funding for our project. So we have not bought licence for Vivado/Vitis. But we do have three ZCU102 boards purchased from Xilinx for our Demo.

**Now results of your previous instructions:**
**Answer for step2:** I followed the step2 of setting up the opencl environment. But I still got the same error.
**Answer for step3:** Attached is the vitis log.
 

Thanks,
Kapoor7997

vt-lib-support, 2022-02-18:
Hi @kapoor7997 ,

Regarding the effect of license expiration, you can get support on the Xilinx forums.

On the build issues:

2. This is Vitis environment setup issue. Installing these [packages](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/acceleration_installation.html#zcy1504034322499__section_t2n_sgj_c2b) (please choose according to your OS) as mentioned in the Vitis documentation page, should install the required OpenCL headers as claimed in the [package details page](https://packages.ubuntu.com/bionic/all/opencl-clhpp-headers/filelist). If it still doesn't install for you, you may need to get the files and manually copy them to your /usr/include/CL directory.
3. From your vits_hls.log it is clear that the error is due to [y2k22 bug](https://support.xilinx.com/s/question/0D52E00006uxy49SAA/vivado-fails-to-export-ips-with-the-error-message-bad-lexical-cast-source-type-value-could-not-be-interpreted-as-target?language=en_US) (This [AR ](https://support.xilinx.com/s/question/0D52E00006vDlvJSAS/export-ip-invalid-argument-revision-number-overflow-issue?language=en_US) covers full info on that). You need patch your Vivado/Vitis installation as explained [here](https://support.xilinx.com/s/article/76960?language=en_US) to overcome it.

vt-lib-support, 2022-02-21:
Hi @kapoor7997 ,

Good to know that you were able to build and generate the sd_card for your project with the base platform.

Before going to the analysis of runtime error, we would like to state that there is no need to update/modify/edit any parts of the generated sd_card image and it is expected to run as is. The [article](https://support.xilinx.com/s/article/1138667?language=en_US) you pointed to is for creating a custom software platform instead of using Xilinx base platform. So it does not apply here.

On the error, it could be an issue with the XRT version. You may try the following; On a new terminal, in the same project folder, source all the environment settings except the xrt and do the host and sd_card build as follows (this forces the executable to use the xrt from sysroot instead of local system):

```
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE = ~/Downloads/XilinxVision/VisionLibraryStuff2020.2/xilinx_zcu102_base_202020_1/xilinx_zcu102_base_202020_1.xpfm
export SYSROOT=~/petali/opencv_build/xilinx-zcu102-2020.2/sysroots/aarch64-xilinx-linux/
export EDGE_COMMON_SW=~/Downloads/XilinxVision/VisionLibraryStuff2020.2/CommonImage2020.2/xilinx-zynqmp-common-v2020.2/
make host sd_card TARGET=hw HOST_ARCH=aarch64
```

kapoor7997, 2022-02-21:
Hi Collaborator
[vt-lib-support](https://github.com/vt-lib-support),

I used all xilinx filea and ran exact commands given byb you.
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE = ~/Downloads/XilinxVision/VisionLibraryStuff2020.2/xilinx_zcu102_base_202020_1/xilinx_zcu102_base_202020_1.xpfm
export SYSROOT=~/petali/opencv_build/xilinx-zcu102-2020.2/sysroots/aarch64-xilinx-linux/
export EDGE_COMMON_SW=~/Downloads/XilinxVision/VisionLibraryStuff2020.2/CommonImage2020.2/xilinx-zynqmp-common-v2020.2/
make host sd_card TARGET=hw HOST_ARCH=aarch64

**And this is the result of the device console after executing the script run_script.sh**

root@zynqmp-common-2020_2:~# 
root@zynqmp-common-2020_2:~# cd /mnt/sd-mmcblk0p1/
root@zynqmp-common-2020_2:/mnt/sd-mmcblk0p1# ls
128x128.png  BOOT.BIN  boot.scr  emconfig.json  krnl_ccm.xclbin    run_script.sh
4k.jpg       Image     ccm       init.sh        platform_desc.txt  system.dtb
root@zynqmp-common-2020_2:/mnt/sd-mmcblk0p1# export XCL_BINDIR=/mnt/sd-mmcblk0p1/
root@zynqmp-common-2020_2:/mnt/sd-mmcblk0p1# sh run_script.sh 
INFO: Running OpenCL section.
[   76.764173] [drm] Pid 1338 opened device
[   76.768120] [drm] Pid 1338 closed device
Found Platform
Platform Name: Xilinx
[   76.778510] [drm] Pid 1338 opened device
[   76.785751] [drm] Pid 1338 closed device
[   76.789781] [drm] Pid 1338 opened device
INFO: Device found - xilinx_zcu102_base_202020_1
XCLBIN File Name: krnl_ccm
INFO: Importing ./krnl_ccm.xclbin
Loading: './krnl_ccm.xclbin'
[   78.096153] [drm] get section DEBUG_IP_LAYOUT err: -22 
[   78.096161] [drm] get section AIE_METADATA err: -22 
[   78.101416] [drm] zocl_xclbin_read_axlf caf49ea2-bc8c-43fb-b2c5-f716414c1f19 ret: 0
[   78.113205] [drm] bitstream caf49ea2-bc8c-43fb-b2c5-f716414c1f19 locked, ref=1
[   78.120886] [drm] No ERT scheduler on MPSoC, using KDS
[   78.133280] [drm] scheduler config ert(0)
[   78.133281] [drm]   cus(1)
[   78.137285] [drm]   slots(16)
[   78.139976] [drm]   num_cu_masks(1)
[   78.142934] [drm]   cu_shift(16)
[   78.146417] [drm]   cu_base(0x80000000)
[   78.149635] [drm]   polling(0)
[   78.153490] [drm] bitstream caf49ea2-bc8c-43fb-b2c5-f716414c1f19 unlocked, ref=0

Thanks,
Kapoor7997

kapoor7997, 2022-02-23:
Today, I built everything from scratch cause I wanted to be sure non of my changes affect the build.
So
I downloaded:- 
1. Fresh copy of Xilinx common image (xilinx-zynqmp-common-v2020.2) 
2. Freshly created a sysroots from xilinx common image by running the sdk.sh script
2. Fresh copy zcu102 base platform

I used the following commands before compiling the **colorcorrectionmatrix** example from **L2**.
source /tools/Xilinx/Vitis/2020.2/settings64.sh
export DEVICE=/home/dev/Downloads/XilinxVision/newtry/xilinx_zcu102_base_202020_1/xilinx_zcu102_base_202020_1.xpfm 
export SYSROOT=/home/dev/Downloads/XilinxVision/newtry/xilinx-zynqmp-common-v2020.2/sysroots/sysroots/aarch64-xilinx-linux/
export EDGE_COMMON_SW=/home/dev/Downloads/XilinxVision/newtry/xilinx-zynqmp-common-v2020.2/
make host sd_card TARGET=hw HOST_ARCH=aarch64

Note: I dit not add any additional flags in the Makefile of colorcorrectionmatrix.

Here is the console output in a text file for the run on hardware(zcu102 board)
[newtry_SameError.txt](https://github.com/Xilinx/Vitis_Libraries/files/8122482/newtry_SameError.txt)

Thanks,
kapoor7997

kapoor7997, 2022-03-02:
Hi [vt-lib-support](https://github.com/vt-lib-support),

As per your suggestion, we tried hw_emu and sw_emu. we ran the following commands from this link(https://xilinx.github.io/Vitis_Libraries/vision/2020.1/overview.html section [](https://xilinx.github.io/Vitis_Libraries/vision/2020.1/overview.html#evaluating-the-functionality)Evaluating the Functionality)

source /tools/Xilinx/Vitis/2020.2/settings64.sh
source /opt/xilinx/xrt/setup.sh 
export DEVICE=/home/dev/Downloads/XilinxVision/newtry/xilinx_zcu102_base_202020_1/xilinx_zcu102_base_202020_1.xpfm 
export SYSROOT=/home/dev/Downloads/XilinxVision/newtry/xilinx-zynqmp-common-v2020.2/sysroots/sysroots/aarch64-xilinx-linux/
make host xclbin TARGET=hw_emu BOARD=Zynq ARCH=aarch64

And 

make host xclbin TARGET=sw_emu BOARD=Zynq ARCH=aarch64

In both cases we used xilinx base image and xilinx common image for sysroot.
Looks like the same erorrs on both cases.

We got multiple errors and logs are attached.
[hw_emu_errors.txt](https://github.com/Xilinx/Vitis_Libraries/files/8171256/hw_emu_errors.txt)
[sw_emu_errors.txt](https://github.com/Xilinx/Vitis_Libraries/files/8171257/sw_emu_errors.txt)

**We have been working on getting the vitis vision libraries working for a month now and haven't gotten anywhgere. 10 days ago, the [vt-lib-support](https://github.com/vt-lib-support) team completely stopped replying. 
We are loosing confidence in xilinx and xilinx products at this point. We are in the process of a multi-million dollar deal and we build our custom hardware using and following the zcu102 board. At this point all base images, base software has failed to provide us with a working demo of the simplest functionality. We would really appreciate some concrete help in this process.**

**I represent Crypto-knight Mining Corp from Vancouver.**

Thanks,
kapoor7997