Xilinx/Vitis_Libraries #83: How to view and modify the API In Vitis_Libraries/data_compression/L2 / tests/ gzipc_block_mm

wanyuzhan, 2021-09-21:
The screenshot below shows the kernel code "gzip_compress_multicore_mm.cpp" in the Vitis_ library calls three APIs to do the compression. But I cannot find where the API internal code is. I would like to know which code implements compression in this project. The introduction of the Vitis accelerator mentions that customers can use Vitis library functions as optimized algorithmic building blocks and modify them. I guess if I want to modify the code, I need to find all the source code first. Or shall I get some suggestions, like what parts can customers modify? 
[image]

Moreover, This screenshot below shows this project's timeline trace. I found I find that some running modules have blank time between them. Is this latency? Or is there any other explanation for this？
[image]

rbhartari-xilinx, 2021-09-21:
Hi @wanyuzhan,

You can find the source code files for all the internal modules of any kernel at below path in repo:
[Vitis_Libraries/data_compression/L1/include/hw/](https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L1/include/hw)

Please refer below image for explanation of timeline trace screenshot.
[image]

The gaps visible are between the subsequent OpenCL API calls in the host code, while host code executes.

heeran-xilinx, 2022-02-02:
closing this issue as no respond to last comment. Please re-open if issue still exist and provide more info to reproduce it.

closed 2022-02-02 by heeran-xilinx