Xilinx/Vitis_Libraries #85: Vitis HLS GUI errors while proj simulation

prkrmx, 2021-09-30:
Hi
I am going through the user guide on how to run a project in the [Vitis HLS GUI](https://xilinx.github.io/Vitis_Libraries/vision/2021.1/overview.html#getting-started-with-hls)
Unfortunately there are errors:

> INFO: [HLS 200-10] Running '/tools/Xilinx/Vitis_HLS/2021.1/bin/unwrapped/lnx64.o/vitis_hls'
> INFO: [HLS 200-10] For user 'max' on host 'idev' (Linux_x86_64 version 5.11.0-37-generic) on Thu Sep 30 22:23:19 IDT 2021
> INFO: [HLS 200-10] On os Ubuntu 20.04.3 LTS
> INFO: [HLS 200-10] In directory '/home/max/rp_ref/vitis_hls/filter'
> Sourcing Tcl script '/home/max/rp_ref/vitis_hls/filter/sobel/solution1/csim.tcl'
> INFO: [HLS 200-1510] Running: open_project sobel 
> INFO: [HLS 200-10] Opening project '/home/max/rp_ref/vitis_hls/filter/sobel'.
> INFO: [HLS 200-1510] Running: set_top sobel_accel 
> INFO: [HLS 200-1510] Running: add_files xf_sobel_accel.cpp -cflags -D__SDSVHLS__ -I../../../../../../../tools/vitis_lib/vision/L1/include -std=c++0x -csimflags -D__SDSVHLS__ -I../../../../../../../tools/vitis_lib/vision/L1/include -std=c++0x 
> INFO: [HLS 200-10] Adding design file 'xf_sobel_accel.cpp' to the project
> INFO: [HLS 200-1510] Running: add_files -tb xf_sobel_tb.cpp -cflags -I../../../../../usr/local/include/opencv4 -I../../../tools/vitis_lib/vision/L1/include -D__SDSVHLS__ -std=c++0x -Wno-unknown-pragmas -csimflags -I../../../tools/vitis_lib/vision/L1/include -D__SDSVHLS__ -std=c++0x -Wno-unknown-pragmas 
> INFO: [HLS 200-10] Adding test bench file 'xf_sobel_tb.cpp' to the project
> INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
> INFO: [HLS 200-10] Opening solution '/home/max/rp_ref/vitis_hls/filter/sobel/solution1'.
> INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
> INFO: [HLS 200-1611] Setting target device to 'xc7z045-ffg900-2'
> INFO: [HLS 200-1505] Using flow_target 'vivado'
> Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-1505.html
> INFO: [HLS 200-1510] Running: set_part xc7z045-ffg900-2 
> INFO: [HLS 200-1510] Running: create_clock -period 10 -name default 
> INFO: [HLS 200-1510] Running: csim_design -ldflags -L/usr/local/lib -lopencv_gapi -lopencv_stitching -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dnn_objdetect -lopencv_dnn_superres -lopencv_dpm -lopencv_highgui -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hfs -lopencv_img_hash -lopencv_intensity_transform -lopencv_line_descriptor -lopencv_quality -lopencv_rapid -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_superres -lopencv_optflow -lopencv_surface_matching -lopencv_tracking -lopencv_datasets -lopencv_text -lopencv_dnn -lopencv_plot -lopencv_videostab -lopencv_videoio -lopencv_xfeatures2d -lopencv_shape -lopencv_ml -lopencv_ximgproc -lopencv_video -lopencv_xobjdetect -lopencv_objdetect -lopencv_calib3d -lopencv_imgcodecs -lopencv_features2d -lopencv_flann -lopencv_xphoto -lopencv_photo -lopencv_imgproc -lopencv_core -argv "/home/max/tools/vitis_lib/vision/data/128x128.png" -clean -quiet 
> INFO: [SIM 211-2] *************** CSIM start ***************
> .
> .
> .
> /lib/x86_64-linux-gnu/libopenmpt.so.0: undefined reference to `std::random_device::_M_getentropy() const@GLIBCXX_3.4.25'
> /usr/local/lib/libopencv_gapi.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
> /usr/local/lib/libopencv_gapi.so: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
> /usr/local/lib/libopencv_gapi.so: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
> /usr/local/lib/libopencv_gapi.so: undefined reference to `std::logic_error::logic_error(std::logic_error&&)@GLIBCXX_3.4.26'
> /usr/local/lib/libopencv_gapi.so: undefined reference to `std::runtime_error::runtime_error(std::runtime_error&&)@GLIBCXX_3.4.26'
> /usr/local/lib/libopencv_dnn.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
> collect2: error: ld returned 1 exit status
> make: *** [Makefile.rules:393: csim.exe] Error 1
> ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
> INFO: [SIM 211-3] *************** CSIM finish ***************

`$ pkg-config --modversion opencv4 >>
4.4.0
`

vt-lib-support, 2021-11-10:
Hi @prkrmx ,

Please try to build the OpenCV-4.4.0 freshly using the compiler from Vitis and use them in your project. See the [OpenCV installation guidance](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#opencv-installation-guidance) for full details.

prkrmx, 2022-07-18:
Works as expected, thanks

closed 2022-07-18 by prkrmx