Xilinx/Vitis_Libraries #90: Corner update function error

wenyee9797, 2021-11-04:
When I call the xf_corner_update.hpp inside the L1 include, the error pop up, it seems like library bug, does anyone meet the same issue?

<-- Error information below -->
caught Tcl error: ERROR: [HLS 207-2276] cannot cast from type 'xf::cv::Mat<3, 360, 640, 1, 2>::_DATATTYPE' (aka 'hls::stream<ap_uint<32>, 2>') to pointer type 'unsigned int *: Vitis_libraries/vision/L1/include/imgproc/xf_corner_update.hpp:31:40

norabarlow, 2021-11-09:
I also encounter the same error.

wenyee9797, 2021-11-10:
I change it to this, the problem solve and works in Vitis flow

for(int i=0;i<height;i++) {
        for(int j=0;j<width;j++) {
            flowvectorsDataPtr[i*width+j] = (unsigned int)flow_vectors.read(i*width+j);
        }
    }

vt-lib-support, 2021-11-10:
Hi @wenyee9797, @norabarlow ,

cornerUpdate is a memory mapped function. Hence, -D__SDA_MEM_MAP__ flag needs to be passed to the compiler. Example [here](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L3/examples/cornertracker/Makefile#L212)

norabarlow, 2021-11-10:
That works, thanks!

closed 2021-11-10 by vt-lib-support