Xilinx/Vitis_Libraries #94: Can't C Synthesis with Quantitive lib bs_model

dajovu2020, 2021-12-02:
Already include bs_model.hpp in Vitis HLS project.
But when we did C Synthesis, there is the error message as below.

ERROR: [HLS 207-812] 'bs_model.hpp' file not found (dut.cpp:18:10)

Are there anything that we need to modfiy?
Thanks.

dajovu2020, 2021-12-02:
The development environment is Vitis HLS 2021.2 in Windows.

We also add the -cflags in Project synthesis setting.
"-DDPRAGMA -I D:/Yuanta_FPGA/Technology/HLS/Example/Vitis_Libraries-master/Vitis_Libraries-master/quantitative_finance/L1/include"

zhenhong-xilinx, 2021-12-10:
I suppose the dut.cpp include path should be "xf_fintech/bs_model.hpp" instead of "bs_model.hpp"

dajovu2020, 2021-12-22:
thanks, it's fixed already.