Xilinx/Vitis_Libraries #98: Demosaicing Makefile fails with check_opencv

nlbutts, 2021-12-12:
I followed the getting started guide: https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/docs/getting-started-with-hls.rst
When I try to make:
```
make run XPART='xcu200-fsgd2104-2-e' CSIM=1
```

This generates the following message:
```
nlbutts@Dell-G15-5515:~/projects/xilinx/Vitis_Libraries/vision/L1/examples/demosaicing$ make run XPART='xcu200-fsgd2104-2-e' CSIM=1
XPART is directly set to xcu200-fsgd2104-2-e
make: *** No rule to make target 'check_opencv', needed by 'setup'.  Stop.
```

I see the check rule in the Makefile, but I can't figure out why it isn't finding the rule.

vt-lib-support, 2021-12-13:
Hello @nlbutts ,

Please try to use the below command to run CSIM, meanwhile we check on the issue:

`make run CSIM=1 DEVICE=<path-to-platform-directory>/< platform >.xpfm `

vt-lib-support, 2022-10-25:
This has been fixed in the latest release. Please reopen if needed.

closed 2022-10-25 by vt-lib-support