Xilinx/Vitis_Libraries #101: ap_shift_reg simulation header file is not applicable for synthesis

haipnh, 2021-12-17:
Hi, 
I'm trying to run the SPARSE example on Alveo U50 with Vitis 2021.1 as following: Vitis_Libraries_2021.1/sparse/L2/tests/fp64.
The host program was built successfully, unless all the kernels (storeYkernel for example):
```
===>The following messages were generated while  performing high-level synthesis for kernel: storeYkernel Log file: /home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/Emulation-SW/build/storeYkernel/storeYkernel/vitis_hls.log :
ERROR: [v++ 207-814] ap_shift_reg simulation header file is not applicable for synthesis: /home/eda/Xilinx-2021.1/Vivado/2021.1/include/ap_shift_reg.h:76:2
ERROR: [v++ 60-300] Failed to build kernel(ip) storeYkernel, see log for details: /home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/Emulation-SW/build/storeYkernel/storeYkernel/vitis_hls.log
ERROR: [v++ 60-773] In '/home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/Emulation-SW/build/storeYkernel/storeYkernel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-814] ap_shift_reg simulation header file is not applicable for synthesis: /home/eda/Xilinx-2021.1/Vivado/2021.1/include/ap_shift_reg.h:76:2
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
makefile:132: recipe for target 'build/storeYkernel.xo' failed
```
I traced the issue caused in the `storeYkernel.tcl` generated by v++ compiler:
```
catch {::common::set_param -quiet hls.xocc.mode csynth};
catch {::common::set_param -quiet hls.enable_scout_hidden_option_error false};
# 
# HLS run script generated by v++ compiler
# 

open_project storeYkernel
set_top storeYkernel
# v++ -g, -D, -I, --advanced.prop kernel.storeYkernel.kernel_flags
add_files "/home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/src/storeYkernel.cpp" -cflags " -g -D SPARSE_dataType=double -D SPARSE_indexType=uint16_t -D SPARSE_parEntries=4 -D SPARSE_hbmChannels=16 -D SPARSE_hbmMemBits=256 -D SPARSE_maxRows=4096 -D SPARSE_maxCols=4096 -D SPARSE_accLatency=8 -D SPARSE_dataBits=64 -D SPARSE_indexBits=16 -I /home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/src -I /opt/xilinx/xrt/include -I /home/eda/Xilinx-2021.1/Vivado/2021.1/include -I /home/haipnh/xilinx_ws/spmv_ws_2021.1/spmv_kernels/include -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/sparse/L1/include/hw -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/sparse/L1/include/sw -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/sparse/L2/include/hw -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/sparse/L2/include/sw -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/sparse/L2/include/hw/fp64 -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1 -I /home/haipnh/repos/Xilinx/Vitis_Libraries_2021.1/blas/L1/include/hw "
open_solution -flow_target vitis solution
set_part xcu50-fsvh2104-2-e
# v++ --hls.clock or --kernel_frequency
create_clock -period 333MHz -name default
# v++ -g or --profile_kernel stall
config_rtl -kernel_profile
# v++ --advanced.param compiler.hlsDataflowStrictMode
config_dataflow -strict_mode warning
# v++ -g
config_debug -enable
# v++ --advanced.param compiler.deadlockDetection
config_export -disable_deadlock_detection=true
# v++ --advanced.param compiler.axiDeadLockFree
config_rtl -m_axi_conservative_mode=1
config_interface -m_axi_addr64
# v++ --hls.max_memory_ports
config_interface -m_axi_auto_max_ports=0
config_export -format ip_catalog -ipname storeYkernel
catch {::common::set_param -quiet hls.enable_synthesis_check_sw_only true};
csynth_design -synthesis_check # <<< This line fails
close_project
puts "HLS completed successfully"
exit
```
What am I doing wrong?

lisaliu1, 2021-12-20:
didn't you built it successfully on U280? If so, you might need to check what you've changed for building it on U50.

haipnh, 2021-12-21:
> didn't you built it successfully on U280? If so, you might need to check what you've changed for building it on U50.

Thank you for you response. I am new to Xilinx so I think investigating compatible projects with my system is better choice.

closed 2021-12-21 by haipnh