Xilinx/Vitis_Libraries #103: Tcl error:  ERROR: [Common 17-55] 'get_property' expects at least one object.

jshashwat, 2021-12-28:
I am building hw_emulation for the gzip L2 demo on Vitis 2021.1. All the kernels are built but make fails at the synthesis stage with the following tcl error.

[09:22:41] Run vpl: Step create_project: Started
Creating Vivado project.
[09:22:43] Run vpl: Step create_project: Completed
[09:22:43] Run vpl: Step create_bd: Started
[09:23:05] Run vpl: Step create_bd: Completed
[09:23:05] Run vpl: Step update_bd: Started
[09:23:05] Run vpl: Step update_bd: Failed
[09:23:06] Run vpl: FINISHED. Run Status: update_bd ERROR
ERROR: [VPL 60-773] In '/home/zero/Documents/Github/Vitis_Libraries-master/data_compression/L2/demos/gzip/_x_temp.hw_emu.xilinx_u200_xdma_201830_2/link/vivado/vpl/runme.log', caught Tcl error:  ERROR: [Common 17-55] 'get_property' expects at least one object.
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, Failed to update block diagram in project required for hardware synthesis.  The project is 'prj'. The update script is '.local/hw_platform/emu/dynamic_post_sys_link.tcl'.  The update script was delivered as part of the hardware platform. An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [09:23:06] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:14 ; elapsed = 00:00:45 . Memory (MB): peak = 1910.250 ; gain = 0.000 ; free physical = 1351 ; free virtual = 26728
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:172: recipe for target 'build_dir.hw_emu.xilinx_u200_xdma_201830_2/compress_decompress.xclbin' failed
make: *** [build_dir.hw_emu.xilinx_u200_xdma_201830_2/compress_decompress.xclbin] Error 1

sw_emulation runs fine for this. Any leads?

heeran-xilinx, 2022-03-15:
Hi @jshashwat ,
Thanks for reporting this issue. but we are not able to reproducible locally and free running kernel support is not there for older platform (2018) for hw_emu flow we disabled the same https://github.com/Xilinx/Vitis_Libraries/blob/master/data_compression/L2/demos/gzip/config.mk#L17
Can you please cross check and if still failing, please provide more details to reproduce the same?

heeran-xilinx, 2022-03-21:
Closing this issue as not a bug as per last comment. Please re-open if still facing the issue and send steps to reproduce.

closed 2022-03-21 by heeran-xilinx