Xilinx/Vitis_Libraries #104: Y2K22 error with revision number in builds

arjunr2, 2022-01-04:
Vitis builds have failed since Jan 1, 2022, possibly due to the format of the stored year for the revision number in the builds. Could someone look into this?

vt-lib-support, 2022-01-04:
Thanks for reporting this.  It has already been reported and is being investigated.  For additional discussion please refer to the following Xilinx Forums Post:

https://support.xilinx.com/s/question/0D52E00006uxYjmSAE/bug-report-vivado-20212-revision-number-overflow-after-new-year?language=en_US

keryell, 2022-01-05:
Do not close issues about problems which still do exist, otherwise people might open them again and again since they will hit the problems.

vt-lib-support, 2022-10-25:
This issue has been resolved:

https://support.xilinx.com/s/article/76960?language=en_US

closed 2022-10-25 by vt-lib-support